BillyBlaze/OctoPrint-TouchUI #145: Menu items not correctly populated

keybkeybkeyb, 2017-08-27:
Hi, My installation of TouchUI shows some visual issues,
- The settings menu that is not populated and few items on the right totally misaligned.
- Items on the other pages are misaligned as well (see attached pics)

I'm using a Raspberry PI2 with a 7" touch screen
Octoprint Version 1.3.4 (master branch)
TouchUI (0.3.8) 
Latest Chromium

The same issue is present with my android device

Hope you can help me.

Thank you,
Alessandro
[image]
[image]
[image]

Ipoc78, 2017-08-28:
I too am having this issue. not only on my tab, cell, but also on web browser. I had to clean my cookies in order to get it back to desktop version on my laptop.

 I'm running  latest version of octpi 0.14 on a raspberry pi3 and using the official pi 7" touch screen (unable to use touch screen due to being hung up on boot up, octprint still works fine)

2tri4fun, 2017-08-29:
I am having similar issues.  Followed your wiki and was able to get touchui up on 7" Pi Touchscreen but control tab is blank.

BillyBlaze, 2017-10-17:
Hello, thanks for reporting and sorry for the long delay.

I think you're having the same issues as #144 and this is currently fixed on the maintenaince branch. However I intend to release 0.3.9 in the next hours where this is fixed.

keybkeybkeyb, 2017-10-17:
Thank you for taking the time for looking into it!

BillyBlaze, 2017-10-17:
0.3.9 has been released, please let me know if this fixed the issue!

Ipoc78, 2017-10-17:
Fantastic thank you for your time and dealing with this issue, 
I'm going to fire up my printer here shortly and report back

Ipoc78, 2017-10-18:
I confirm on my end that it's working correctly

BillyBlaze, 2017-10-18:
Awesome! Glad this was the issue that is now resolved 👍 

Thanks again for reporting! Closing this, please open a new one if needed!

closed 2017-10-18 by BillyBlaze