BillyBlaze/OctoPrint-TouchUI #146: OOP,s Something went wrong

storemanuk, 2017-08-29:
I keep getting red notifications in the octoprint web interface stating.....

Something went wrong
Please consult octoprint.log for details but there are no Octoprint logs to consult. 

These notifications popup all the time even if touchui is not being used. They also popup immediately when the octoprint server starts up. I have tried to uninstall so I can re-install but even though the Octoprint plugin manager says the uninstall was successful, touch ui isn't uninstalled - no, it simply pops up the 'something went wrong notification. The same thing happens when I try to disable touchui from within plugin manager.

Looks like the only option left open to get rid of these annoying popup messages is to reflash the SD card with a clean Octoprint.
[image]

storemanuk, 2017-08-29:
Immediately after posting, I connected to the printer through Octoprint and straightaway got 3 popup notifications
[image]

BillyBlaze, 2017-08-29:
Sounds like there is an issue with PIP, @foosel do you have any idea where to start? Manual remove the folder from the venv?

foosel, 2017-08-29:
If it pops up directly on start I'm not positive it's a pip issue.
JavaScript error console and octoprint.log might be interesting to dig
deeper here. I've never seen the notification in the second screenshot
before...

To manually uninstall a plugin see
https://github.com/foosel/OctoPrint/wiki/FAQ#i-installed-a-broken-plugin-how-do-i-uninstall-it-again

Starting in safe mode might also be interesting:
http://docs.octoprint.org/en/master/features/safemode.html

>

BillyBlaze, 2017-08-29:
Thanks!

The second notification: TouchUI tries to compile LESS to CSS and store through an API call into a static file. This can only be [done by an admin](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/octoprint_touchui/api.py#L20) and the response of the API is currently forbidden.

storemanuk, 2017-08-29:
Thanks to foosel. I uninstalled using the instructions from the link you gave so no longer get the popups in firefox when switching pages, loading gcode to print or any other action unrelated to using a mobile device.

Will get round to installing again soon and see what happens. Don't know if it is related but when I uninstalled I was prompted to upgrade pip though this issue with the notifications only started happening yesterday. Was running quite happily with the older pip version up to that point.

BillyBlaze, 2017-10-17:
Glad you figured out the issue @storemanuk!

I am closing this issue, please let me know if this happens again

closed 2017-10-17 by BillyBlaze