BillyBlaze/OctoPrint-TouchUI #147: Question about 3rd party plugin support (navbar items)

ntoff, 2017-08-30:
Is you manually adding workarounds for navbar plugins the only way to get them to play nicely with touch ui?

[image]

Or is there a way of doing this in the troublesome plugin? The plugin isn't registered so I have no idea how many people actually use it (I discovered it in one of octoprint's issues threads), nor is it originally mine, so I don't want to request you add a workaround for it (I made some tweaks to it so the version I have installed isn't even the same as the original one), but is there a way for 3rd party plugins to put themselves in the right place without you having to constantly add workarounds?

p.s. I'm pretty newb at this so if this is an obvious question, I apologise.

BillyBlaze, 2017-08-30:
Hi,

Thanks for reporting/asking. yeah currently I am manually moving some plugins. I could ofcourse add an extra check to see if the items are a link or not. In case it's not I could move them to the bottom section in the dropdown.

However I am not sure if this will always work, if HTML is defined differently I still might need to apply some styles to make it appear correct.

I'll try to see if I can add this in the next release.

p.s. do you have a link to that plugin so I can test it out?

ntoff, 2017-08-30:
Here's the original https://github.com/larp-welt/OctoPrint-FanSpeed

but I've messed about with it (for some reason he made it an html link when all it is, is a piece of text with no menu) https://github.com/ntoff/OctoPrint-FanSpeed which is why I was curious if I could do it myself since I'm using a different version to the original.

Here's the thread where I originally found the plugin: https://github.com/foosel/OctoPrint/issues/1187#issuecomment-316219754 original author kind of hints that he/she made it just for themselves and we're free to use it if we want but it's not registered in the repo so there's probably a slim chance that there's only like 2 or 3 of us even using it (so is it worth a workaround?).

BillyBlaze, 2017-10-17:
This has been fixed in the current release 0.3.9

Thanks for reporting!

closed 2017-10-17 by BillyBlaze