BillyBlaze/OctoPrint-TouchUI #152: Steps 5 and 6 not working

Budoman, 2017-09-22:
I've followed all of the steps successfully up to the end of step 5. My pi touchscreen is connected with the dsi ribbon cable, not hdmi, so I assumed I was to use the optional code snippet. I saved my 99-fbdev.conf file. Then I ran into trouble. I'm a noob, so I figured I would cut and paste ls "/dev | grep fb" in the Mac OS Terminal to check the touchscreen mount name, but it didn't work, so I skipped over that part. I ran the command in step 6 and Terminal returned the following: "sed: can't read /etc/X11/Xwrapper.config: No such file or directory".

So now what? I'm stuck.

calabris, 2017-09-23:
I am having the same problem when I run the command to let anyone run it. I get /etc/X11/Xwrapper.config No such file or directory.  My Pi boots to the desktop, but no browser launches.  I cannot even manually launch Chromium

BillyBlaze, 2017-09-24:
Hello,

You both can ignore step 6.
On step 5, did you try using fb0 or fb1 when using fb0?

BillyBlaze, 2017-10-10:
Any luck?

BillyBlaze, 2017-10-17:
Closing after inactivity, if you have any info please post them here.

closed 2017-10-17 by BillyBlaze