BillyBlaze/OctoPrint-TouchUI #154: [PLUGIN CONFLICT] With 'Custom Background' plugin...

bolsoncerrado, 2017-10-02:
...upon uploading a new background, TouchUI shows BOTH the old and the new at the same time, see here:

https://thingiverse-production-new.s3.amazonaws.com/renders/fe/12/c1/92/a6/33ce92a1055b1e49cac26ad7748a2753_preview_featured.jpg

jneilliii, 2017-10-05:
Billy, does touch ui apply the background-image css property to a different element vs a default octoprint install?

bolsoncerrado, 2017-10-05:
@BillyBlaze that is :)

BillyBlaze, 2017-10-05:
@bolsoncerrado, thank you for reporting.

@jneilliii, yes, it looks like I move the [background image to a parent](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/source/less/tabs/temperature.less#L18). It has been a while, so I can't remember why I did that. Lemme first check if I can move this background definition back to the same element. Since this issue is more TouchUI related.

BillyBlaze, 2017-10-17:
This has been fixed in the current release 0.3.9

Thanks for reporting!

closed 2017-10-17 by BillyBlaze

bolsoncerrado, 2017-10-19:
Seems we have a complete mix instead now @BillyBlaze  
https://i.imgur.com/BBQmVIZ.png