BillyBlaze/OctoPrint-TouchUI #155: [HELP REQUEST!] Can't get TouchUI to start (stays on login prompt)

bolsoncerrado, 2017-10-05:
Hi @BillyBlaze 

I've been using oPi and TouchUi for centuries and so far all bugs were ironed as things progressed.

I'm now on a Pi3 w/jessie and the official Pi 7" touch, and using an .IMG to create new SD cards as I get new printers in.

Today's SD works fine, Pi boots fine, display works fine but all I get is the login screen. I've searched past issues related to staying on the login screen, to no avail, so to put things into perspective:

- already done aptget update/upgrade
- no errors found on octoprint.log
- already reinstalled touchui n times now
- tried to reinstall chromium (and applying the above afterwards, no joy)
- tried to install legacy xorg settings
- xinit ~/TouchUI-autostart/chromium.xinit brings this:

----------------------------------------------------------------------------------
----------------------------------------------------------------------------------
X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux chappie 4.4.27-v7+ #2 SMP Wed Oct 26 19:09:12 UTC 2016 armv7l
Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0x92f2f783 smsc95xx.macaddr=B8:27:EB:F2:F7:83 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:VGA8x8
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
Current version of pixman: 0.33.3
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Thu Oct  5 16:41:43 2017
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
----------------------------------------------------------------------------------
----------------------------------------------------------------------------------

- $ chromium-browser --debug brings this:
----------------------------------------------------------------------------------
----------------------------------------------------------------------------------
# Env:
#     LD_LIBRARY_PATH=/usr/lib/chromium-browser:/usr/lib/chromium-browser/libs
#                PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
#            GTK_PATH=
# CHROMIUM_USER_FLAGS=
#      CHROMIUM_FLAGS=  --enable-pinch --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0
/usr/bin/gdb /usr/lib/chromium-browser/chromium-browser -x /tmp/chromiumargs.rduBfv
GNU gdb (Raspbian 7.7.1+dfsg-5+rpi1) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/lib/chromium-browser/chromium-browser...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/lib/chromium-browser/chromium-browser --enable-pinch --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x74440290 (LWP 1630)]
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:1005/161922.716200:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[New Thread 0x73aff290 (LWP 1638)]
[1627:1627:1005/161922.755453:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[1627:1627:1005/161922.852257:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display: 
[Thread 0x74440290 (LWP 1630) exited]
[Thread 0x75122000 (LWP 1627) exited]
[Inferior 1 (process 1627) exited with code 01]
-------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------

Any clues?

Thanks!

BillyBlaze, 2017-10-05:
Hi @bolsoncerrado,

Can you run ```sudo xinit ~/TouchUI-autostart/chromium.xinit``` instead of ```xinit ~/TouchUI-autostart/chromium.xinit```?

bolsoncerrado, 2017-10-05:
(EE) 
Fatal server error:
(EE) no screens found(EE) 
(EE) 

WTF?!?!?

bolsoncerrado, 2017-10-05:
- 99-fbdev.conf:

Section "Device"
  Identifier "myfb"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection

bolsoncerrado, 2017-10-05:
Hmmmm 

/dev $ ls fb*
fb0

bolsoncerrado, 2017-10-05:
HOWEVER the fully working pi/touch printer I have on its side has also fb0 on /dev and /dev/fb1 on 99-fbdev.conf ?!?!??!?!?

bolsoncerrado, 2017-10-05:
OK just saw this one: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/44

So I should change fb1 to fb0 on the fbdev conf file.....but WHY the other Pi works fine!?!?!?

bolsoncerrado, 2017-10-05:
OK no joy either. Changed to fb0, nothing happened.

BillyBlaze, 2017-10-05:
can you post the whole output of  ```sudo xinit ~/TouchUI-autostart/chromium.xinit```?

bolsoncerrado, 2017-10-05:
pi@chappie:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit?

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux chappie 4.4.27-v7+ #2 SMP Wed Oct 26 19:09:12 UTC 2016 armv7l
Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0x92f2f783 smsc95xx.macaddr=B8:27:EB:F2:F7:83 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:VGA8x8
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
Current version of pixman: 0.33.3
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Oct  5 18:16:17 2017
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) no screens found(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.

BillyBlaze, 2017-10-05:
It looks like xinit cannot find the file ```/usr/share/X11/xorg.conf.d/99-fbdev.conf``` and continues looking in other folders for configurations.

What is the output of ```ls -la /usr/share/X11/xorg.conf.d/``` ???

bolsoncerrado, 2017-10-05:
pi@chappie:~ $ ls -la /usr/share/X11/xorg.conf.d/ 
total 24
drwxr-xr-x 2 root root 4096 Mar 16  2017 .
drwxr-xr-x 5 root root 4096 Mar 16  2017 ..
-rw-r--r-- 1 root root 1099 Sep 24  2016 10-evdev.conf
-rw-r--r-- 1 root root 1350 Nov 11  2016 10-quirks.conf
-rw-r--r-- 1 root root 1753 Jun  1  2016 70-synaptics.conf
-rw-r--r-- 1 root root   98 Apr  5  2017 99-fbdev.conf
pi@chappie:~ $

bolsoncerrado, 2017-10-05:
OK!
Got it back to work!

By editing the 99-fbdev file and setting to fb0 instead of fb1 it now works!

So now the questions:

1) Differences between /usr/share/X11/xorg.conf.d folder and /etc/X11/xorg.conf ???
2) Why fb0 if the touchui says fb1 !?!?

BillyBlaze, 2017-10-05:
Ok I'm glad you fixed it!!

1. /usr/share/ is for program configs
    /etc/X11/ is for system configs

   both will be scanned by xserver

2. You misread or either jumped a important sections in the guide:

    > If your touchscreen is connected with HDMI, paste the following code:
    > *Use this if you're using the Official Pi Touchscreen*

bolsoncerrado, 2017-10-05:
Yeah but what I mean is I never realized a out the /usr folder for. X11 before, and the printer that works. Fine points to. Fb1. I wonder why it. Works. On that one.

closed 2017-10-10 by BillyBlaze