BillyBlaze/OctoPrint-TouchUI #156: TouchUI wont start

tntx250, 2017-10-09:
hello i am trying to start touchUi for 2 days now i tried tons of stuff and reinstall octopi 3 times now. 
At the moment i get this error when i try to run xinit ~/TouchUI-autostart/chromium.xinit
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
/home/pi/TouchUI-autostart/chromium.xinit: 34: /home/pi/TouchUI-autostart/chromium.xinit: xinput: not found
TouchUI: touch events automatically detected
TouchUI: screensaver enabled
matchbox-window-manager: error while loading shared libraries: libpango-1.0.so.0: cannot open shared object file: No such file or directory
/usr/lib/chromium-browser/chromium-browser: error while loading shared libraries: libpango-1.0.so.0: cannot open shared object file: No such file or directory
xinit: connection to X server lost

i tried to install the libraries with no luck

any1 know whats going wrong?

BillyBlaze, 2017-10-10:
Hi tntx250,

Could you please provide a detailed log? You can find them under `/var/log/Xorg.0.log` (use cat to output the contents)

tntx250, 2017-10-10:
Hello. it manage to lunch after 3-4 re-installs, maybe a mirror with some libs was not functioning correctly.
I didnt do anything specific and i cant post fixes for someone that may got this errors. 
Thanks for your work on that plugin its awsome!

BillyBlaze, 2017-10-17:
Glad to hear the problem is solved! Kinda weird, but lets hope it was a incident.
Your welcome and enjoy it! 👍

closed 2017-10-17 by BillyBlaze