BillyBlaze/OctoPrint-TouchUI #158: Connecting to TouchUI failed

AndreasHeisinger, 2017-10-11:
Hi,

i had installed a Octoprint by using the fresh Octoprint-Image on an SD-Card. I had installed a 7 Inch Raspberry TFT.
Then i had installed TouchUI like the Manual Boot to Browser Jessie Light. Everthing works . 
If i reboot the System runs up, then the TFT shows TouchUI connecting and after a while the Text became red and i see a Message "Connecting to TouchUI failed".
I had installed like this before on a 3.5 Inch NonRaspberry TFT and it works fine, but working with the 3.5 inch display is horrible so i took the 7 inch Display. It doesn't work. Can you help me?
Thank you very Much!!!

AndreasHeisinger, 2017-10-11:
Hi,
it works. I finalized the Installation in Octoprint and add the TouchUI Plugin and it works. 
Sorry for my Question!
Andreas

Penged, 2017-10-17:
Andreas, How did you finalize your installation? I am having the same issue as you and could use your help.

BillyBlaze, 2017-10-17:
@AndreasHeisinger, glad you figured out the issue.

@Penged, can you share your octoprint.log?

BillyBlaze, 2017-10-18:
@Penged, please create an issue if you need help.

Closing this issue

closed 2017-10-18 by BillyBlaze