BillyBlaze/OctoPrint-TouchUI #160: Issue with 7 inch touchscreen

Penged, 2017-10-18:
I installed  Octoprint by using the fresh Octoprint-Image on an SD-Card. I had installed a 7 Inch Raspberry TFT.
Then i had installed TouchUI like the Manual Boot to Browser Jessie Light. Everthing works .
If i reboot the System runs up, then the TFT shows TouchUI connecting and after a while the Text became red and i see a Message "Connecting to TouchUI failed".
   This issue was on a different thread that was resolved by finalizing the installation of octoprint. I was asked to share my octoprint.log before I was asked to create a new post. I am not sure how to finalize the installation or how to find my octoprint.log. Sorry I am not more skilled in this. Any help would be greatly appreciated.

BillyBlaze, 2017-10-19:
Hi @Penged,

Have you installed TouchUI within the OctoPrint plugin manager?

If not;
1. Open octoprint webinterface
1. Goto the Settings
1. Goto the Plugin Manager
1. Click on "Install more" on the bottom
1. Search for TouchUI and install it 👍

Penged, 2017-10-19:
Well I have made progress. I have managed to get it to start up on the TFT screen but it does not seem to connect to my printer. I have a Rostock max V2 with a stock rambo board.

ticals1000, 2017-10-20:
you have to sign in on the touchscreen, open the menu on the top right then enable virtual keyboard, then click on sign in and use your octopi account to sign in

Penged, 2017-10-20:
Yes, I have got it working. One last question. Can I run the touchscreen without having octoprint started on my laptop?

ticals1000, 2017-10-20:
Of course

Sent from my Verizon, Samsung Galaxy smartphone

terracedr, 2017-11-06:
Penged "Well I have made progress. I have managed to get it  to start up on the TFT screen but it does not seem to connect to my  printer. I have a Rostock max V2 with a stock rambo board."
----------------------------------------------------------------------------------------------------
Wondering how you fixed the problem of Octopint loaded with out TouchUI? I now am having the same issue after a new image disk with Octoprint and TouchUi is installed in the plugins of the web inter face of Octoprint. However, I cannot get the RPU 7" inch screen to load the TouchUI. PLEASE HELP>

Penged, 2017-11-06:
terracedr, 
    I dont rememeber how I got it to work. I tried startting my printer and then the touchscreen, or vice versa. I did connect after a few tries. I In the end I scrapped the project. It was more effective to start prints off my laptop through octoprint

ticals1000, 2017-11-06:
You have to follow the directions for the "boot to browser" using raspbian jessie light after you have "already installed" touchui.   https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light) The only thing that is not in the directions though is that you have to create the xwrapper.config first in /etc/X11/Xwrapper.config.  otherwise it looks like everything goes through fine but it does not

Sent from my Verizon, Samsung Galaxy smartphone

BillyBlaze, 2017-12-12:
Closing this issue since the OP resolved it and is not using it anymore.

For sake of my sanity, please create a new issue if you're having issues.

closed 2017-12-12 by BillyBlaze