BillyBlaze/OctoPrint-TouchUI #161: If screen is rotated from landscape to portrait the menu icons gets stacked on top of each other.

Linusoderlind, 2017-10-20:
In the latest release the icons show if they fit in the menu. But they are not removed if I rotate my phone from landscape to portrait.

I use a Nexus 5x with Android 8.0.0 and the latest Chrome 61.0.3163.98
The images are from reloading the screen in portrait, rotating to landscape and back to portrait.
[image]
[image]
[image]

I can reproduce the behavior on my Windows 10 PC with Chrome 62.0.3202.62 if I resize the window small enough so some icons are moved to the drop down. I then maximize the window and then un maximize it again. If I now resize the window by dragging the GUI behaves again.

BillyBlaze, 2017-10-21:
Hi @Linusoderlind 

Thanks for reporting and for providing the detailed information/screenshots.
I understand where this goes wrong and will fix this in the upcoming release!

Linusoderlind, 2017-10-21:
Working as a software tester has it's perks. :)

commit 9090f100 (2017-12-12): :bug: Aggressive resize

Be more aggressive on trigger resize functions for mobile devices.

Fixes #161
closed 2017-12-12 by BillyBlaze