BillyBlaze/OctoPrint-TouchUI #163: iOS Safari Login Box Issue

Sunoo, 2017-10-25:
On Safari on iOS, when you tap the Login option, the menu expands but the login UI elements do not appear. Scrolling the screen usually seems to resolve the issue.

veggie2u, 2017-10-27:
This is not just Safari iOS, but I have duplicated the issue there as well.  I have a pi running epiphany fullscreen with an Adafruit touch screen. I think everything else is OK, but login has issues with that configuration also. If I scroll first, then press the login item I can then see the name/password fields. Part of the issue is the limited size of the screen on the PiTFT.

BillyBlaze, 2017-12-12:
Thanks both for reporting!

@Sunoo, the issue has been found and the fix is currently pushed to maintenance and will be released very soon.

@veggie2u, Epiphany support have been dropped for a while now (mostly because of the bad performance and lack of some important standards). I suggest upgrading to Chromium and installing the TouchUI bootloader as instructed on my wiki.

commit d64f1134 (2017-12-12): :bug: Fix iOS rendering

Add `backface-visibility: hidden;` to the login dropdown to force
iOS safari to render the dropdown menu.

Fixes #163
closed 2017-12-12 by BillyBlaze