BillyBlaze/OctoPrint-TouchUI #164: octoprint loaded without touchui

krea002, 2017-11-05:
hi i am having an issue with touchui not booting on the local touch screen.  yesterday everything was working well and after i finished my print i shut don the system when i booted it up to day i get an error.  Octoprint loaded without Touchui tap to retry.  i have retried several time s with no luck.

I then tried to uninstall the touchui plug in no luck.
i checked for updates only to find that the touchui is up to date.
i uninstalled all plugins to ensure they where not causing an issue and reinstalled touchui only, no luck.

BillyBlaze, 2017-11-22:
Hi @krea002,

Can you provide me your OctoPrint log?

BillyBlaze, 2017-12-04:
Any updates? if not I'm closing this issue.

rcbadiale, 2017-12-08:
Hi, I am having the same problem, only tried to update though.

Here goes an Octoprint log:
[octoprint (1).log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/1541109/octoprint.1.log)

BillyBlaze, 2017-12-08:
Lets dive a bit deeper in the Chromium log.
Can you enable that?

1. Stop TouchUI `sudo service touchui stop`
1. Run ```nano ~/TouchUI-autostart/chromium.xinit```
1. At line 10 remove the hashtag ```#TOUCHUI_DEBUG="--remote-debugging-port=9222 --enable-logging --v=100"```
1. Save & Exit (CTR-O, Enter, CTR-X)
1. Start TouchUI `sudo service touchui start`
1. Share the new log after you received the error found at ```/home/pi/.config/chromium/chrome_debug.log```

BillyBlaze, 2017-12-08:
Can you try this out to see if it works again?

1. Stop TouchUI:
````
sudo service touchui stop
````

2. Remove default user in Chrome
````
rm -rf ~/.config/chromium/Default
rm -rf ~/.cache/chromium
````

3. Start TouchUI
````
sudo service touchui start
````

BillyBlaze, 2017-12-10:
Hi @rcbadiale,

Glad this fixed it, I will release in the upcoming release that TouchUI will always load if you're using TouchU with the autostart script to prevent this from happening again.

@krea002 does this also fix it for you?

BillyBlaze, 2017-12-12:
I just released a new version of TouchUI, and in this version I will always enable TouchUI if it's getting loaded with the bootloader. This should prevent issues like this occurring.

closed 2017-12-12 by BillyBlaze

maganzi, 2019-01-08:
Hey i have the same issue. But this will don´t work for me.

[image]

[image]

Please help me.

> Can you try this out to see if it works again?
> 
> 1. Stop TouchUI:
> 
> ```
> sudo service touchui stop
> ```
> 1. Remove default user in Chrome
> 
> ```
> rm -rf ~/.config/chromium/Default
> rm -rf ~/.cache/chromium
> ```
> 1. Start TouchUI
> 
> ```
> sudo service touchui start
> ```

BillyBlaze, 2019-01-08:
Althrough the symptoms look the same, you've a different issue. The new ForceLogin plugin is not working with the out-dated bootloader. Try this steps, if that is not working please create a new issue:

1. [Update your bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader)
1. Run `sudo ~/TouchUI-autostart/helpers/install`
1. Setup autoLoginAs during the installation.

MrArmageddon, 2019-01-13:
Was having the same issue, and updating the bootloader, then setting the Auto Login fixed it. Thanks!

putanmihai2006, 2019-02-25:
> 
> 
> Can you try this out to see if it works again?
> 
>     1. Stop TouchUI:
> 
> 
> ```
> sudo service touchui stop
> ```
> 
>     1. Remove default user in Chrome
> 
> 
> ```
> rm -rf ~/.config/chromium/Default
> rm -rf ~/.cache/chromium
> ```
> 
>     1. Start TouchUI
> 
> 
> ```
> sudo service touchui start
> ```

Hey. How do you actually go from getting stuck on that "octopi loaded without touchui" window to a terminal?