BillyBlaze/OctoPrint-TouchUI #165: 0.3.10 Virtual Keyboard Broken

lciscon, 2017-11-06:
In the latest update the virtual keyboard isn't working properly anymore.  When you pop up the keyboard the display area for the value field is much smaller than it used to be and when you click on the keys nothing displays in that area.  When you press keys a value is stored, but because there is no way of seeing what is being entered it is almost impossible to enter a correct value.  

You can duplicate the issue easily by turning on TouchUI in a standard browser window and enable the popup keyboard.

Can we get this fixed asap?  This is a critical bug.

BillyBlaze, 2017-11-22:
Thanks for reporting!

Yes, I will fix this as soon as I can get my hands on some code.

BillyBlaze, 2017-12-01:
I cannot reproduce your issue (see screenshot below). Have you tried changing the font-size within the TouchUI settings? Maybe this problem will resolve itself once you recompile a clean CSS file.

[image]

lciscon, 2017-12-01:
It is only an issue in the numeric target temperature field (as far as I know).  The other text input popups work correctly.

[image]

BillyBlaze, 2017-12-01:
Ahh yes, found that here too now! Thanks for reporting, fix is currently on maintenance branch and will be released at +/- the same time as octoprint 1.3.6

lciscon, 2017-12-04:
Perfect thanks.  If you cold also double check in this fix that when you enter a value from the keyboard that the temperature actually starts changing? I noticed that even if I happen to get a correct value in with the broken interface I can't get it to trigger an actual temperature change.  But if I pick ABS from the pulldown it works correctly.

BillyBlaze, 2017-12-04:
Will do, just asking for clarification; You type with your keyboard inside the virtual keyboard and the value change doesn't happen after finishing it?

lciscon, 2017-12-04:
You type it in and the value displays in the field but the actual command to change the temperature doesn't get sent to the printer.  So the hotend temperature never actually changes.

BillyBlaze, 2017-12-12:
Thanks for reporting this too! This will also be fixed in the upcoming release.

commit 6f8e0bf3 (2017-12-12): :bug: Do not prevent double-click on temp

OctoPrint now uses form submits on temperature input changes.

Instead of disable the submit button - which prevents submitting -
exlude the button from double-click prevention

Fixes #165
closed 2017-12-12 by BillyBlaze