BillyBlaze/OctoPrint-TouchUI #166: Webcam is either laggy or non existent in some cases with autostart script

amoose136, 2017-11-13:
When setting up a webcam if the address is set to `http://<local ip>/webcam/?action=stream` the webcam will work locally but not when connected remotely over the internet because of course the scope of the ip is wrong for a remote host. If I instead put `http://<url resolving to global ip>/webcam/?action=stream` it will work in all cases but be laggy because now all webcam traffic (!) is routed over the web even for local connections. If I instead put simply `/webcam/?action=stream` it works in all cases and with minimum lag except when Touchui is loaded locally via the touchui-autostart script. There is no issue if the chromium instance is launched manually. What is the autostart script doing differently than just launching a chromium instance? Why can't a chromium instance launched this way resolve `/webcam/?action=stream`?

BillyBlaze, 2017-11-22:
Hi @amoose136,

The problem is that I made the autostart files agnostic of platform (i.e. OctoPi or bare Jessie), 
The autostart files extract the port number of OctoPrint and directly connects to the localhost as followed: `http://localhost:5000/`

However the webcam is available on port 80, and not on port 5000, so the relative url will be `http://localhost:5000/webcam/?action=stream` and that doesn't exist.

After thinking a bit of it, and looking at the startup code, you could might fix this another way, since the startup files load TouchUI default settings after OctoPrint, we might just overwrite the port number there. (This will not affect OctoPrint)

1. Stop TouchUI
    ````
    sudo service touchui stop
    ````

1. Edit /etc/default/touchui
    ````
    sudo nano /etc/default/touchui
    ````

1. Add the following:
    ````
    port=80
    ````

1. Save and exit

1. Start TouchUI
    ````
    sudo service touchui start
    ````

amoose136, 2017-11-28:
That seems to have fixed it.

closed 2017-11-28 by amoose136