BillyBlaze/OctoPrint-TouchUI #168: Can't get past login

siblues, 2017-11-19:
I have tried setting up Boot to browser using this guide and I can't seem to get it working.I am also having a really hard time because of my lack of experience or knowledge.Everything goes as expected and explained in this how-to until I get to step 5.I paste the following string into my SSH window

`sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf`

I then paste 
`Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection`

I save and exit but I am assuming that if after everything is done if I re-enter

`sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf`

It should have what was pasted into it still there but it is empty.Any help would be appreciated.

BillyBlaze, 2017-11-22:
Hi @siblues,

Have you [expanded your file system](https://www.raspberrypi.org/documentation/configuration/raspi-config.md)?

siblues, 2017-11-22:
I have everything working now but every time I touch the TFT it interupts the current print and the printer turns off.The PSU control switches the relay off everytime I try to use the touchscreen.I am guessing that you can only use the TouchUi or the Octoprint browser page by themselves?

BillyBlaze, 2017-11-23:
That sounds like a hardware issue. Everything connected correctly?

siblues, 2017-11-23:
Yeah everything is working as it should but I am using three gpio's and so is the TFT so I am really new to this but that is probably the problem.I may try to use GPIO pins 33,35,and 37 so that the TFT and relay aren't using the same GPIO's.

BillyBlaze, 2017-11-23:
Yeah, that sounds like a culprit.
Good luck on resolving this.

closed 2017-11-23 by BillyBlaze