BillyBlaze/OctoPrint-TouchUI #170: Accidently turned off touchui on my touch display now it wont start

supernielsen, 2017-11-22:
I tried to be as accurate as i could in the title.

I have a 5 inch touch screen on my pi everything have been working perfect, however i accidently turned off touch ui on the display, i didnt care that day and closed everything as normal. 

Then when i started the printer today it says.

                    TOUCHUI
   Octoprint loaded without TouchUI 
                  Tap to retry

No matter how many times i retry it wont work. what to do?

BillyBlaze, 2017-11-22:
Hi @supernielsen,

Hmm, I guess I have created a soft-block there 🤣 

The easy way to fix this is to reset Chrome:

1. Disable TouchUI:
````
sudo service touchui stop
````

2. Remove default user in Chrome
````
rm -rf ~/.config/chromium/Default && rm -rf ~/.cache/chromium
````

3. Start TouchUI
````
sudo service touchui start
````

BillyBlaze, 2017-12-04:
Any updates? if not I'm closing this issue.

supernielsen, 2017-12-04:
so sorry..!!!

Yes it worked a charm thank you so much ! :)

BillyBlaze, 2017-12-04:
No problem and awesome it works again 👍

closed 2017-12-04 by BillyBlaze

BillyBlaze, 2017-12-12:
Btw. I just released a new version of TouchUI, and in this version I will always enable TouchUI if it's getting loaded with the bootloader. This should prevent issues like this occurring.

supernielsen, 2017-12-13:
I actually just got that update yesterday so thats just perfect.. :D