BillyBlaze/OctoPrint-TouchUI #171: move left right buttons don't work on the Touch Screen

henrikkoch, 2017-11-26:
All the buttons but left/right work when using TouchUI on my Prusa i3MK2s/MM 
It works fine form the web browser (not touch panel) But only left/right buttons on the Touch screen doen't work.

Any ideas why?  How can this be changed?

I'm using OctoPrint Version 1.3.5 (rc/maintenance branch)

TouchUI (0.3.10)

BillyBlaze, 2017-11-27:
Hi @henrikkoch,

TouchUI doesn't manipulate the underlying functions of this functionality. So OctoPrint should handle this.

If you press the left/right can you see any activity in your terminal output?

henrikkoch, 2017-11-29:
Hi BillyBlaze.  When trying up/down I get Z commands like:  Send: G1 Z10 F200

When trying ahead/back i get commands like:  Send: G1 Y10 F6000

Nothing happens when pressing the left/right buttons?

BillyBlaze, 2017-11-29:
Do you see the left/right/home button changing colors when clicking them? Perhaps you have a 3rd party plugin that overlaps these controls?

Also I see you're on an older version of 1.3.5, you should be able now to install the stable version of 1.3.5 now. (e.g. master branch)

henrikkoch, 2017-12-01:
No - They doesn't seem to change colors when you pushed these buttons. But the others yes

henrikkoch, 2017-12-01:
Did update to OctoPrint: 1.3.6rc1   Still same problem. No color change on mentioned buttons and no printer functionality

henrikkoch, 2017-12-04:
BillyBlaze any other ideas how to get my setup to work?

BillyBlaze, 2017-12-04:
Hmm, for me it works on any test device, do you have any 3rd party plugins installed?

BillyBlaze, 2017-12-12:
Closing this issue due to lack of information.

closed 2017-12-12 by BillyBlaze