BillyBlaze/OctoPrint-TouchUI #174: [Feature Request] disable camera tab

ManuelW77, 2017-12-02:
Hey there,
i have a wish for touch ui. I want to disable the camera tab, and loadings, because it takes a lot of cpu on my pi and i don't need it.

So it would be nice if it could be possible to disable it in settings.

Regards
Manuel

BillyBlaze, 2017-12-03:
Hi Manuel,

If you want to save CPU then you need to disabling the webcam program itself. The webcam program generates a jpeg image on your Pi and that gets loaded into OctoPrint/TouchUI, so your CPU wont be affected by disabling it in OctoPrint/TouchUI.

You need to disable `mjpg-streamer` and then you can disable the webcam tab in itself by removing your webcam configuration inside the OctoPrint settings.

TouchUI wont show a camera tab once it's disabled.

ManuelW77, 2017-12-03:
Hey Billy,
on my Pi 3 touchui (with chromium) uses a lot of cpu, not the mjpg-streamer process.
[image]

BillyBlaze, 2017-12-04:
Wow, that is alot of CPU usage.

Do you have any extra plugins installed?

BillyBlaze, 2017-12-04:
That are alot of plugins, they should be loaded into TouchUI.

Can you try disabling these plugins and try again if you have such high CPU usage:
1. STL Viewer
2. Print History

ManuelW77, 2017-12-04:
I disabled booth plugins, but nothing changed. CPU usage is the same as before.

ManuelW77, 2017-12-04:
I disables mostly all plugins and can get the cpu usage a bit down to a max of 90%.
I think there is something other wrong.

BillyBlaze, 2017-12-04:
Hmm, this is weird. So does the CPU drop at all? or is it hanging always at +/- 90% now?

Also what version of chromium are you using?

ManuelW77, 2017-12-04:
Yes, the CPU is dropping also. Going up and down, but while printing it's between 60-120%.

Chromium 60.0.3112.89 Built on Ubuntu 14.04, running on Raspbian 9.1

BillyBlaze, 2017-12-04:
Thanks for the info! I needed to upgrade some things here in order to see if I can reproduce it.  I am currently stuck on Jessie with Chromium 56. So we know atleast that was stable.

I have a feeling that Chromium 60 on Stretch is not stable yet.

BillyBlaze, 2017-12-15:
Sorry I still haven't got around to build a new develop environment with Jessie Stretch.

However as I was thinking about this issue, could you check if you're running in touchscreen mode? That mode should disable some CPU heavy things.

mmontminy, 2018-12-30:
I came here looking for a solution to this, as I'm seeing the same behavior. At first I just assumed chromium was a cpu pig on pi, but when playing around with repetier server, I noticed they also use chromium for the ui, and it sips cpu.

As with the OP, I tried removing most plugins with no real impact. While it's not affected printing, it's a concern that it might.

Per your last comment, I did check that chrome isn't being forced to touchscreen mode via /etc/defaults/touchui. If it's being forced otherwise, I can't say, but I don't see any command line args being passed to chromium that would do so.

I know this is a pretty old issue, but I'm hoping someone's found a solution to this.

jph290, 2019-01-14:
@mmontminy  I am also seeing the same thing.  Just setup TouchUI with the official raspberry pi touchscreen.  Chromium has gone up to 134% (must be more than 1 cores worth).

ksztibor, 2019-03-30:
I also experience the high cpu usage (about 120%) of chromium occasionally, when touchui is used.

Adrian-at-CrimsonAuzre, 2019-05-22:
I'd like to ask if we can disable _any_ tab when TouchUI is enabled. If I'm mounting this to my printer, I also don't need to access the Timelapse or Octolapse tabs for example.

stepman0, 2020-03-17:
Anything new on this issue. I have a similar problem with high load due to camera - which is not needed next to the printer. But I need the camera for remote access...

TheRealAlexV, 2020-05-17:
This is also a problem for me. I have an rpi4 with the official 7" touchscreen. When the touchui is open chromium runs at around 130%.

@BillyBlaze is this an issue still on your radar?

closed 2020-12-09 by github-actions[bot]