BillyBlaze/OctoPrint-TouchUI #175: Cannot get touch ui to boot to browser, goes to login prompt

DBaker85, 2017-12-05:
So I have been trying for a couple of days to get this working but I just cannot. 
These are the steps I followed:

### Setup [lcdshow](https://github.com/goodtft/LCD-show)

this results in this message: 
`pi@octopi: sudo ./LCD35-show
   cp: cannot create regular file "/usr/share/X11/xorg.conf.d/": Not a directory
   this is not raspberrypi kernel, no need to update touch configure, reboot`

It works though

### Install touch ui
This works

### Follow install steps
`sudo sed -i 's/allowed_users=console/allowed_users=anybody/' /etc/X11/Xwrapper.config` 
This step fails

### Reboot:
Everything works but I stay stuck on the login prompt no black or white screen

### Addtitional step
edit `/etc/X11/xorg.conf.d/99-fbdev.conf`

Still nothing but octoprint works fine on network

I have no idea what to do now. Any help would be greatly appreciated

BillyBlaze, 2017-12-05:
Hi,

Can you share the contents of ```/var/log/Xorg.0.log```

DBaker85, 2017-12-05:
Here you go,  Also forgot to mention I am using a pi3 and a 3.5inch gpio xpt2046 screen

[    12.236]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    12.236] X Protocol Version 11, Revision 0
[    12.236] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    12.236] Current Operating System: Linux octopi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l
[    12.236] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_$
[    12.236] Build Date: 11 November 2016  11:59:59AM
[    12.236] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    12.237] Current version of pixman: 0.33.3
[    12.237]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    12.237] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    12.237] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Dec  5 12:08:30 2017
[    12.311] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    12.311] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    12.479] (==) No Layout section.  Using the first Screen section.
[    12.479] (==) No screen section available. Using defaults.
[    12.480] (**) |-->Screen "Default Screen Section" (0)
[    12.480] (**) |   |-->Monitor "<default monitor>"
[    12.509] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    12.509] (**) |   |-->Device "touchscreen"
[    12.509] (**) |   |-->GPUDevice "touchscreen"
[    12.509] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    12.509] (==) Automatically adding devices
[    12.509] (==) Automatically enabling devices
[    12.509] (==) Automatically adding GPU devices
[    12.539] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    12.548] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    12.548]    Entry deleted from font path.
[    12.548] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    12.548]    Entry deleted from font path.
[    12.548] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    12.548]    Entry deleted from font path.
[    12.548] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    12.548]    Entry deleted from font path.
[    12.549] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    12.549]    Entry deleted from font path.
[    12.549] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    12.549]    Entry deleted from font path.
[    12.549] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    12.549]    Entry deleted from font path.
[    12.549] (==) FontPath set to:
        built-ins
[    12.549] (==) ModulePath set to "/usr/lib/xorg/modules"
[    12.549] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    12.549] (II) Loader magic: 0x54cd8f20
[    12.549] (II) Module ABI versions:
[    12.549]    X.Org ANSI C Emulation: 0.4
[    12.549]    X.Org Video Driver: 20.0
[    12.549]    X.Org XInput driver : 22.1
[    12.549]    X.Org Server Extension : 9.0
[    12.553] (--) using VT number 2

[    12.553] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    12.553] (II) no primary bus or device found
[    12.558] (II) LoadModule: "glx"
[    12.598] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    12.878] (II) Module glx: vendor="X.Org Foundation"
[    12.878]    compiled for 1.18.4, module version = 1.0.0
[    12.878]    ABI class: X.Org Server Extension, version 9.0
[    12.878] (==) AIGLX enabled
[    12.878] (II) LoadModule: "fbdev"
[... 50 lines cut ...]
[    13.056] (==) FBDEV(0): DPMS enabled
[    13.064] (==) RandR enabled
[    13.121] (II) SELinux: Disabled on system
[    13.132] (II) AIGLX: Screen 0 is not DRI2 capable
[    13.132] (EE) AIGLX: reverting to software rendering
[    13.132] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so: cannot open shared object file: $
[    13.132] (EE) GLX: could not load software renderer
[    13.132] (II) GLX: no usable GL providers found for screen 0
[    13.208] (EE) Error compiling keymap (server-0)
[    13.208] (EE) XKB: Couldn't compile keymap
[    13.208] (EE) XKB: Failed to load keymap. Loading default keymap instead.
[    13.236] (EE) Error compiling keymap (server-0)
[    13.237] (EE) XKB: Couldn't compile keymap
[    13.237] XKB: Failed to compile keymap
[    13.237] Keyboard initialization failed. This could be a missing or incorrect setup of xkeyboard-config.
[    13.237] (EE)
Fatal server error:
[    13.237] (EE) Failed to activate core devices.(EE)
[    13.237] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    13.237] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    13.237] (EE)
[    13.246] (EE) Server terminated with error (1). Closing log file.

DBaker85, 2017-12-07:
I am about to give up on this. should I just install the desktop env and launch it from there?
I think the pi 3 should be able to handle it being multi core and all...

BillyBlaze, 2017-12-07:
Sorry, I have no idea whats going on, I can't reproduce your issue. Building a full jessie will make it easier to setup xserver. since that is pre-configured.

Just checking some stuff I found other websites with the same error:
```
[ 13.237] (EE) XKB: Couldn't compile keymap
[ 13.237] XKB: Failed to compile keymap
[ 13.237] Keyboard initialization failed. This could be a missing or incorrect setup of xkeyboard-config.
[ 13.237] (EE)
Fatal server error:
[ 13.237] (EE) Failed to activate core devices.(EE)
```

Do you have enough disk left? Did you install anything else except LCD-show driver?

DBaker85, 2017-12-07:
I did not install anything else no. I have an 8gb card. Might be a little light, but should be enough right?

BillyBlaze, 2017-12-08:
This is a bizar error I have never encountered before or have been reported before. 

Did you try to rebuild it from scratch? Perhaps something went wrong during setup with dependencies.

If your reluctant todo this, you could try to see what the outcome is of the follow steps:
1. Check your disk space with `df -h`
2. `ls ~/.Xauthority -al`
2. `ls /tmp/ -al`
2. `ls /var/tmp/ -al`

DBaker85, 2017-12-08:
### 1.
 Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.3G  1.7G  5.3G  25% /
devtmpfs        458M     0  458M   0% /dev
tmpfs           462M     0  462M   0% /dev/shm
tmpfs           462M  6.3M  456M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           462M     0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1   41M   21M   21M  51% /boot

### 2
`ls: cannot access /home/pi/.Xauthority: No such file or directory`

### 3
total 32
drwxrwxrwt  8 root root 4096 Dec  8 21:10 .
drwxr-xr-x 22 root root 4096 Jun 21 09:49 ..
drwxrwxrwt  2 root root 4096 Dec  8 21:09 .font-unix
drwxrwxrwt  2 root root 4096 Dec  8 21:09 .ICE-unix
drwxr-xr-x  3 pi   pi   4096 Dec  8 21:10 pip-7C1bEn-build
drwxrwxrwt  2 root root 4096 Dec  8 21:09 .Test-unix
drwxrwxrwt  2 root root 4096 Dec  8 21:10 .X11-unix
drwxrwxrwt  2 root root 4096 Dec  8 21:09 .XIM-unix

### 4
total 8
drwxrwxrwt  2 root root 4096 Dec  8 21:10 .
drwxr-xr-x 11 root root 4096 Jun 21 09:49 ..

I am not against rebuilding it, i have in fact tried and lost many hours of my life.. I am not that good at linux so yeah... Also to mention I tried a waveshare hdmi screen also and could not get it to work...

I just found a 16gb card. Should I try that?

DBaker85, 2017-12-10:
Well it seems your instinct was correct. 
I found a 16gb card (Lucky find!) and redid all the steps and it worked perfectly the first time. 
I have no idea if the space was the problem or if the card itself was faulty. 

Well I suppose now you can add a section to the wiki that says to try a different/bigger card if you have weird issues.

Thanks for your time and effort to help me out!

BillyBlaze, 2017-12-10:
Phew! Glad that you where able to make it work.

Everything else looked okay, you had enough disk space, all rights look good, I just have no idea what caused this, and to be frankly I don't even wanna know at this stage because debugging apt-get would be hell.

Your welcome and enjoy 👍

closed 2017-12-10 by BillyBlaze