BillyBlaze/OctoPrint-TouchUI #177: Cant initiate a print

cwem3d, 2017-12-27:
With my RPI 3 w/ octoprint and touch ui I can not get it to print it works fine on my phone but not the PI.
https://youtu.be/cVToxRrn2Gw
OctoPrint 1.3.6 on OctoPrint 0.14.0

BillyBlaze, 2017-12-27:
Hi, Thanks for reporting. It seems I cannot reproduce your issue at the moment.

Can you provide some more information with the Octoprint log and if there is any error thrown inside the browser console?

The get the browser console, you might be better off to navigate with Chrome to OctoPrint on a computer and extract it from there.

cwem3d, 2017-12-28:
Where exactly can I find the logs in Windows?

BillyBlaze, 2017-12-29:
should be under `%APPDATA%/OctoPrint`

cwem3d, 2017-12-30:
There is no octoprint folder in apdata.

BillyBlaze, 2017-12-30:
sorry, I misunderstood you, OctoPrint is installed on your RPI, so the log should be there under `~/.octoprint/log` on your Pi.

BillyBlaze, 2017-12-30:
in addition you can also download the log from OctoPrint interface under `Settings > Logs`

cwem3d, 2018-01-01:
https://drive.google.com/file/d/12YssES9e72BdiIBg6ARFMPA6AJZXB2Um/view?usp=sharing

cwem3d, 2018-01-01:
Could it be do to me using m33 fio with a M3D

BillyBlaze, 2018-01-01:
Yeah, M33 Fio is a very bad plugin that breaks OctoPrint when running it on a raspberry Pi.

Since the plugin is not using the standards of OctoPrint, I am unable to prevent it from loading on a Pi.

cwem3d, 2018-01-01:
Will M33 FIO be supported by your plugin further down the road?

BillyBlaze, 2018-01-01:
Not by me, it's unworkable and I would lose my sanity and jump out of the window.

On a serious note; I would be willing to accept pull-requests but I will not actively develop for that plugin. It just does to many things on multiple non-octoprint way. For example; it directly executes javascript and doesn't follow the 'octoprint way'. It's impossible to stop it from loading and cleaning up the mess would be alot of time I don't even have to start with.

closed 2018-01-23 by stale[bot]