BillyBlaze/OctoPrint-TouchUI #178: Connetcting to TouchUI failed

Lleldor, 2017-12-30:
Hi,

my Pi zero W always failed to connect to Touch UI. It works perfect on all devices, except on the 2.8" touch display on my Pi zero.  I tried 3 times a complete fresh installation, without any success. It always says: connecting to touch UI for several minutes and then prompting Connecting failed. 

On my Mac, iPhone, iPad I can switch from Touch Ui to the common interface and back without any issues. It's just the touch display. :)

BillyBlaze, 2017-12-30:
Thanks for reporting. Can you provide me your OctoPrint.log?

Lleldor, 2017-12-30:
Wow, thanks for the quick reply! 
Is it this, what you need?

2017-12-30 14:03:06,022 - octoprint.startup - INFO - Blacklist processing done
2017-12-30 14:03:06,507 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-12-30 14:03:24,940 - octoprint.plugin.core - INFO - Found 11 plugin(s) providing 11 mixin implementations, 10 hook handlers
2017-12-30 14:03:26,152 - octoprint.server - INFO - Intermediary server started
2017-12-30 14:03:26,161 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-12-30 14:03:29,577 - octoprint.plugin.core - INFO - Found 11 plugin(s) providing 11 mixin implementations, 10 hook handlers
2017-12-30 14:03:30,002 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-12-30 14:03:30,015 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-12-30 14:03:31,204 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-12-30 14:03:56,321 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2017-12-30 14:04:20,972 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-12-30 14:04:20,998 - octoprint.plugin.core - INFO - Initialized 11 plugin implementation(s)
2017-12-30 14:04:21,119 - octoprint.plugin.core - INFO - 11 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  Fan Speed Control (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fanspeedslider
|  LED Strip Control (0.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  TouchUI (0.3.11) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2017-12-30 14:04:36,019 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 1
|    freq: 1000.0
|    ram: 388681728
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: Zero W
|      revision: 9000c1
|      version: 0.14.0
|  python:
|    pip: 9.0.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2017-12-30 14:04:36,054 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2017-12-30 14:04:36,175 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2017-12-30 14:04:36,965 - octoprint.server - INFO - Shutting down intermediary server...
2017-12-30 14:04:36,997 - octoprint.server - INFO - Intermediary server shut down
2017-12-30 14:04:37,006 - octoprint.events - INFO - Processing startup event, this is our first event
2017-12-30 14:04:37,012 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2017-12-30 14:04:37,023 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-12-30 14:04:37,032 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-12-30 14:04:37,110 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2017-12-30 14:04:37,164 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2017-12-30 14:04:37,182 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2017-12-30 14:04:37,185 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2017-12-30 14:04:37,368 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-12-30 14:04:38,124 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid

BillyBlaze, 2017-12-30:
hmm that looks good, what logs are all available under `~/octoprint/logs`?

Lleldor, 2017-12-30:
hmm ...
none.

Lleldor, 2017-12-30:
Alright, seems to be a timing issue.
This is what happened:
After start up, everything is fine on all devices, except the touch display. It stated "Connecting to Touch UI" failed. 
I tapped on the display and it changed to: "Loading Octoprint", and after more than 5min to: "Loading TouchUI". 
Additional 3- 5 min of waiting, and TouchUI shows up on the display as well. 
During all this waiting time everything is fine and working on the other devices.

BillyBlaze, 2018-01-01:
I have updated the bootloader, if you [update yours](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) does it work then?

p.s. I wouldn't recommend the Zero W with a touchscreen attached, performance is horrible to be honest.

Lleldor, 2018-01-02:
Thanks,

I'll try as soon as I'm back.
I agree. The Pi zero might not be the best option. It's running my Octopi for month now, and I bought a display for a really low price in China. Why not using it? Should be better than no display at all. I don't expect to get a video stream from the camera. In fact I don't need that, when I'm standing in front of the printer. :)
For starting a print job, or pausing one, it should be good enough. (I hope).

BillyBlaze, 2018-01-02:
Yeah, I was planning to use a tablet with TouchUI as control point and the Zero W only for printing communication and hosting TouchUI.

But it turns out that on higher printing speeds that the Zero W will be a bottleneck. So I am upgrading mine to Pi3.

closed 2018-01-23 by stale[bot]