BillyBlaze/OctoPrint-TouchUI #179: No TouchUI image on Waveshare 3.5" with 2017-11-29-octopi-stretch-lite-0.15.img

kimnach, 2018-01-06:
First off, I am just starting out in the 3D-printing arena (Folgertech FT-5--still unused).  Because I have a spare RPi with a Waveshare SpotBear (3.5" touch, https://www.waveshare.com/3.5inch-rpi-lcd-a.htm), which had been used with HomeGenie for home automation, I decided that before firing up the FT5 I would use the RPi with TouchUI.

The installation and set-up went fine with the Octopi Jessie-lite (version 0.14).  The TouchUI interface displayed on the 3.5" screen.  But being one to not leave well-enough alone, I saw that there are nightly builds available with Stretch.  So, I downloaded 2017-11-29-octopi-stretch-lite-0.15 and proceeded to install it on a 32GB Samsung EVO.  All went well until TouchUI.  There is something which I did subtley differently or there is an issue with.....well, probably user error.

I followed the instructions, as I had with the Jessie version, as given at https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light).  All that happens is the arrow first displays in the center, an under-score cursor then briefly in the upper right, the arrow flashes again in the center of the 3.5" screen, then the screen goes blank.  Undaunted, I checked out this thread and went completely through it: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/98.

Issuing a chromium-browser command results only in "[1547:1547:0106/010757.905273:ERROR:browser_main_loop.cc(582)] Failed to put Xlib into threaded mode.  
(chromium-browser:1547): Gtk-WARNING **: cannot open display:"

lightdm is up to date (also ran update, upgrade, and dist-upgrade all in hopes of solving issue).

here is my Xorg log:
[Xorg.0.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/1608343/Xorg.0.log)

Fortunately I have the image of the working Jessie version to which I can always revert.  In the mean time, I will image this and then try to a full raspbian-Stretch with octoprint and TouchUI.

BillyBlaze, 2018-01-06:
Hi @kimnach,

Thanks for reporting, first off I cannot see any obvious issues at all.
Are you sure that the driver you installed (LCD-show) is available for Stretch?? Or was it a version for Jessie?

The issue #174 that was reported, was someone who successfully set up TouchUI on Stretch. So if you have the right driver, it should be do-able.

Althrough I am still unsure that Chromium is stable on Stretch. See #174 for some details.

kimnach, 2018-01-06:
Paul,
I just installed the full version of Stretch, booted, and installed the video driver for the 3.5" monitor (again, it plugs into the GPIO).  The monitor and touch work, so it wasn't a driver issue.  I will see if I have time tonight to install Octoprint.

BTW, one of the oddities that occured yesterday (after my post) was that the touchui displayed on my windoze box's browser when Octoprint server started.  I forget which of the settings turned it off.

Thanks,
greg
http://kimnach.org

kimnach, 2018-01-08:
Paul, I know that this is a different issue than what I originally had posted, and probably would be best addressed in a new ticket or having this one re-titled.  I cannot get Octoprint server to start.  I tried late into the wee hours of the morning to install Octoprint (per https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian) --first with a virgin installation of a full Raspbian Stretch and then with subsequent updates and then dist-upgrade.  Each time I copied the commands line-by-line into PuTTy.

During the initial installation of Octoprint, starting the server (PuTTy side) froze at "writing cache" or some such echo; however, the Raspian desktop remains responsive on the 3.5" LCD--so the pi is not freezing up.  Rebooting and attempting to start the server freezes at different points.  I have tried numerous times to start the server and, of course, it is not accessible from a browser.  I will note that during the installtion process (for example at "**./venv/bin/python setup.py install**") there were WARNINGS having to do with various missing .gif, .au, .txt, folders already either exist or do not exist, etc.  However, there never were any ERRORS echoed back.  Example:
	Running future-0.15.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-	gyqWBn/future-0.15.2/egg-dist-tmp-AV6Lx6
	warning: no files found matching '*.au' under directory 'tests'
	warning: no files found matching '*.gif' under directory 'tests'
	warning: no files found matching '*.txt' under directory 'tests'
	zip_safe flag not set; analyzing archive contents...

Sunday, 7 January 2017 Install Attempt
Today I decided to attempt install as root to see whether or not that helps.  Well, still no joy.  Raspbian Stretch boots into the desktop and displays on the 3.5" LCD.  But Octoprint seems not to be properly installed.

Not knowing which log files would help, here is a step-by-step output of commands when logged in as root.  (After the inability to connect to Octoprint with a browser and desperate for some success, I "over-wrote" the installation by re-entering all of the commands while logged in as "pi" and prefixing all commands with "sudo."  See end of post: "OVER RE-INSTALL.")

Command lines are shown, and the echos (as appropriate) follow.

root@raspberrypi:~# apt-get install python-pip python-dev python-setuptools python-virtualenv git libyaml-dev build-essential
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
git is already the newest version (1:2.11.0-3+deb9u2).
python-dev is already the newest version (2.7.13-2).
python-dev set to manually installed.
python-setuptools is already the newest version (33.1.1-1).
python-setuptools set to manually installed.
python-pip is already the newest version (9.0.1-2+rpt1).
The following additional packages will be installed:
  python3-virtualenv virtualenv
Suggested packages:
  libyaml-doc
The following NEW packages will be installed:
  libyaml-dev python-virtualenv python3-virtualenv virtualenv
0 upgraded, 4 newly installed, 0 to remove and 36 not upgraded.
Need to get 187 kB of archives.
After this operation, 513 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian-us.ngc292.space/raspbian stretch/main armhf libyaml-dev ar             mhf 0.1.7-2 [48.9 kB]
Get:2 http://raspbian-us.ngc292.space/raspbian stretch/main armhf python-virtual             env all 15.1.0+ds-1 [61.1 kB]
Get:4 http://raspbian-us.ngc292.space/raspbian stretch/main armhf virtualenv all              15.1.0+ds-1 [19.7 kB]
Get:3 http://mirrors.syringanetworks.net/raspbian/raspbian stretch/main armhf py             thon3-virtualenv all 15.1.0+ds-1 [57.7 kB]
Fetched 187 kB in 21s (8,714 B/s)
Selecting previously unselected package libyaml-dev:armhf.
(Reading database ... 113509 files and directories currently installed.)
Preparing to unpack .../libyaml-dev_0.1.7-2_armhf.deb ...
Unpacking libyaml-dev:armhf (0.1.7-2) ...
Selecting previously unselected package python-virtualenv.
Preparing to unpack .../python-virtualenv_15.1.0+ds-1_all.deb ...
Unpacking python-virtualenv (15.1.0+ds-1) ...
Selecting previously unselected package python3-virtualenv.
Preparing to unpack .../python3-virtualenv_15.1.0+ds-1_all.deb ...
Unpacking python3-virtualenv (15.1.0+ds-1) ...
Selecting previously unselected package virtualenv.
Preparing to unpack .../virtualenv_15.1.0+ds-1_all.deb ...
Unpacking virtualenv (15.1.0+ds-1) ...
Setting up libyaml-dev:armhf (0.1.7-2) ...
Setting up python3-virtualenv (15.1.0+ds-1) ...
Setting up python-virtualenv (15.1.0+ds-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up virtualenv (15.1.0+ds-1) ...

root@raspberrypi:~# git clone https://github.com/foosel/OctoPrint.git
Cloning into 'OctoPrint'...
remote: Counting objects: 42836, done.
remote: Total 42836 (delta 0), reused 0 (delta 0), pack-reused 42836
Receiving objects: 100% (42836/42836), 31.73 MiB | 4.22 MiB/s, done.
Resolving deltas: 100% (27521/27521), done.

root@raspberrypi:~# cd OctoPrint
root@raspberrypi:~/OctoPrint# virtualenv venv
[... 91 lines cut ...]

pi@raspberrypi:~/OctoPrint $ ~/OctoPrint/venv/bin/octoprint serve
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/bin/octoprint", line 6, in <module>
    from pkg_resources import load_entry_point
[... cut ...]
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'argh>=0.24.1' distribution was not found and is required by watchdog

This just muddied the waters.

Luckily I have the Jessie-Lite version to fall back on.
Thanks,
greg
http://kimnach.org

BillyBlaze, 2018-01-14:
Hi Greg,

Thanks for the detailed information!
And I'm glad you where able to make it work.

I will use your information once I'll get to supporting Stretch.
For now I am closing this issue.

Thanks again and regards,
Paul

closed 2018-01-14 by BillyBlaze