BillyBlaze/OctoPrint-TouchUI #180: Chromium on running on Raspberry Pi 3 & 7” Raspberry Pi LCD wont launch

Braxstonwithans, 2018-01-08:
My chromium browser only worked for a split second after the initial installation. Now the browser will not open at all. The icon is there and when I click on it I get a loading cursor.

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000 dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9b19e749-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 November 2016 11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
Before reporting problems, check http://wiki.x.org
to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
(++) from command line, (!!) notice, (II) informational,
(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Mon Jan 8 20:54:26 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
(EE)
(EE)
Please consult the The X.Org Foundation support
at http://wiki.x.org
for help.
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.

BillyBlaze, 2018-01-09:
Hi @Braxstonwithans,

It seems like TouchUI is getting executed by the Pi user, and this should be root.
Could you tell me where you followed the instructions?

Braxstonwithans, 2018-01-09:
Originally, I followed the instructions from the Octopi website then I followed the instructions from Adafruit [https://youtu.be/thQ3PdLxqe8]. After neither processes worked out, I began to troubleshoot and I stumbled into the recommendations where I originally posted the issue. Any suggestions?

BillyBlaze, 2018-01-09:
Ahh I see, those are sadly outdated tutorials.

Please follow the up-to-date instructions from https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

Start from the beginning and do not skip steps. You can skip installing drivers since the Official RaspberryPi Touchscreen comes pre-installed on OctoPi 0.13.

Braxstonwithans, 2018-01-09:
I tried incorporating those steps before as well via [http://mrarmageddon.blogspot.com/2017/03/3d-printing-octopi-setup-with-7in-pitft.html](url). This is what I'm getting.

chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0109/150053.727673:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[8394:8394:0109/150053.751756:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[8394:8394:0109/150053.772902:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display: 
pi@octopi:~ $ /usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd
-bash: /usr/lib/chromium-browser/chromium-browser:: No such file or directory
pi@octopi:~ $ ~/TouchUI-autostart 
-bash: /home/pi/TouchUI-autostart: Is a directory
pi@octopi:~ $ /etc/default/touchui
-bash: /etc/default/touchui: Permission denied
pi@octopi:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@octopi:~ $ sudo apt-get install rpi-update
Reading package lists... Done
Building dependency tree       
Reading state information... Done
rpi-update is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@octopi:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date

BillyBlaze, 2018-01-10:
Can you please share the log found under `/var/log/Xorg.0.log`

Run `cat /var/log/Xorg.0.log` to see it's contents.

Braxstonwithans, 2018-01-11:
pi@octopi:~ $ cat /var/log/Xorg.0.log
[     6.471] 
X.Org X Server 1.18.4
Release Date: 2016-07-19
[     6.471] X Protocol Version 11, Revision 0
[     6.471] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[     6.471] Current Operating System: Linux octopi 4.9.75-v7+ #1074 SMP Mon Jan 8 14:24:50 GMT 2018 armv7l
[     6.471] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9b19e749-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[     6.471] Build Date: 11 November 2016  11:59:59AM
[     6.472] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
[     6.472] Current version of pixman: 0.33.3
[     6.472]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     6.472] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     6.472] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Jan  9 17:17:05 2018
[     6.480] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     6.492] (==) No Layout section.  Using the first Screen section.
[     6.492] (==) No screen section available. Using defaults.[     6.492] (**) |-->Screen "Default Screen Section" (0)
[     6.492] (**) |   |-->Monitor "<default monitor>"
[     6.498] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[     6.498] (**) |   |-->Device "touchscreen"
[     6.498] (**) |   |-->GPUDevice "Allwinner A10/A13 FBDEV"
[     6.498] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[     6.498] (==) Automatically adding devices
[     6.498] (==) Automatically enabling devices
[     6.498] (==) Automatically adding GPU devices
[     6.504] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     6.514] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     6.514]    Entry deleted from font path.
[     6.514] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     6.514]    Entry deleted from font path.
[     6.514] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     6.514]    Entry deleted from font path.
[     6.514] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[     6.515]    Entry deleted from font path.
[     6.515] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     6.515]    Entry deleted from font path.
[     6.515] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     6.515]    Entry deleted from font path.
[     6.515] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[     6.515] (==) ModulePath set to "/usr/lib/xorg/modules"
[     6.515] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     6.515] (II) Loader magic: 0x69df20
[     6.515] (II) Module ABI versions:
[     6.515]    X.Org ANSI C Emulation: 0.4
[     6.515]    X.Org Video Driver: 20.0
[     6.515]    X.Org XInput driver : 22.1
[     6.515]    X.Org Server Extension : 9.0
[     6.517] (--) using VT number 2

[     6.518] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     6.518] (II) no primary bus or device found
[     6.519] (II) LoadModule: "glx"
[     6.534] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     6.653] (II) Module glx: vendor="X.Org Foundation"
[     6.653]    compiled for 1.18.4, module version = 1.0.0
[     6.653]    ABI class: X.Org Server Extension, version 9.0[     6.653] (==) AIGLX enabled
[     6.653] (II) LoadModule: "fbdev"
[     6.653] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[     6.657] (II) Module fbdev: vendor="X.Org Foundation"
[     6.657]    compiled for 1.18.4, module version = 0.4.4
[     6.657]    Module class: X.Org Video Driver
[     6.657]    ABI class: X.Org Video Driver, version 20.0
[... 54 lines cut ...]
[     8.235] (II) config/udev: Adding input device FT5406 memory based driver (/dev/input/event0)
[     8.235] (**) FT5406 memory based driver: Applying InputClass "libinput touchscreen catchall"
[     8.235] (II) LoadModule: "libinput"
[     8.235] (II) Loading /usr/lib/xorg/modules/input/libinput_drv.so
[     8.282] (II) Module libinput: vendor="X.Org Foundation"
[     8.282]    compiled for 1.18.4, module version = 0.20.0
[     8.282]    Module class: X.Org XInput Driver
[     8.282]    ABI class: X.Org XInput driver, version 22.1
[     8.282] (II) Using input driver 'libinput' for 'FT5406 memory based driver'
[     8.282] (**) FT5406 memory based driver: always reports core events
[     8.282] (**) Option "Device" "/dev/input/event0"
[     8.282] (**) Option "_source" "server/udev"
[     8.285] (II) input device 'FT5406 memory based driver', /dev/input/event0 is tagged by udev as: Touchscreen
[     8.298] (II) input device 'FT5406 memory based driver', /dev/input/event0 is a touch device
[     8.350] (**) Option "config_info" "udev:/sys/devices/virtual/input/input0/event0"
[... cut ...]
[     8.351] (**) FT5406 memory based driver: (accel) acceleration threshold: 4
[     8.351] (II) input device 'FT5406 memory based driver', /dev/input/event0 is tagged by udev as: Touchscreen
[     8.352] (II) input device 'FT5406 memory based driver', /dev/input/event0 is a touch device
[     8.352] (II) config/udev: Adding input device FT5406 memory based driver (/dev/input/mouse0)
[     8.352] (II) No input driver specified, ignoring this device.
[     8.352] (II) This device may have been added with another device file.

Braxstonwithans, 2018-01-14:
Any luck?

sanfordsteve, 2018-01-14:
I am having the similar issue.  I followed the instructions on the wiki.  Chromium does not open although its icon is present, and is selectable from the menu. It simply does not open.  I have installed epiphany-browser and am able to open and use the TouchUI interface.  Wonder if there is an issue with Chromium and Pi3

BillyBlaze, 2018-01-14:
I don't see any issues in the logs. I am also confused by what you guys are reporting. How do you mean; you click on the icon? 

Did you guys install PIXEL desktop? and installed the TouchUI start-up scripts?

sanfordsteve, 2018-01-15:
Raspberry Pi Desktop  "sudo ./scripts/install-desktop"

BillyBlaze, 2018-01-15:
So thats a yes? You installed the desktop and touchui?

sanfordsteve, 2018-01-17:
Yes, both desktop and touchui
Touchui does work on my Android tablet as well as the epiphany browser.
Again I wonder if there is an issue with chromium working on the Pi?

Steve

BillyBlaze, 2018-01-17:
No, there is no issue with Chromium. It's how you set up your OS.

The TouchUI-autostart files is a replacement for the desktop. Since we only want to run OctoPrint and TouchUI there is no reason to load the whole PIXEL interface. Both the desktop and TouchUI-autostart will be launched with "Xserver".

However, you're now creating two instances of "Xserver", one that is invisible and contains only Chromium and the another one that is visible is the Desktop. Chromium has been opened in the invisible frame, so all the interactions will be inside that frame.

What can you do?
1. Determine what you want, boot to desktop? or boot to chrome?
2. Disable either the desktop or TouchUI-autostart

Disabling Desktop:
1. Run `systemctl set-default multi-user.target`
2. Reboot

Disabling TouchUI-autostart:
1. Run `sudo nano /etc/default/touchui`
2. Change `START=yes` to `START=no`
3. Save and exit (CTR-O, enter, CTR-X)
4. Reboot

closed 2018-02-07 by stale[bot]