BillyBlaze/OctoPrint-TouchUI #182: Pi Camera works on remote but not with Official 7" LCD

overpaul, 2018-01-23:
TouchUI is being run on pi 3B with the official 7" touch screen and pi camera
Clean install
The web camera shows up with Octoprint in a web browser in either full Octoprint or TouchUI mode.
The web camera shows up with TouchUI on an android tablet and phone
The web camera does NOT show up on the pi driven 7" touchscreen that has the pi camera attached.
Playing with web camera settings, per the blog instructions got it to work on the touchscreen, but then it did not work on the web or Android.
I think the issue is on the pi with the 7" touchscreen (the one the camera is attached to) you needs a local address (127.0.0.1) but that does not work remotely.
Perhaps the solution would be to allow a different streaming address on the pi with the camera attached from the one that remote touchUI and Octoprint uses. 
Or just take the camera option off when run locally as you probably don't need it when you are at the printer.

BillyBlaze, 2018-01-23:
Hi,

Thanks for reporting, Please have a look at this solution: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/166#issuecomment-346404542

overpaul, 2018-01-23:
> Thanks for reporting, Please have a look at this solution: #166 (comment)

Thanks for the quick response.

Fixed the problem.
I did an internet search before contacting you and didn't find this
suggestion.

You might want to add it to the posting on "Setup: Boot to Browser (OctoPi
or Jessie Light)"
I am sure other's may have the same issue and the webcam troubleshooting
guide takes you in a different direction to solve the problem.

Thanks

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

BillyBlaze, 2018-02-01:
Your welcome 👍 

I do not promote this work-around and I need to fix this issue soon on a more robust way, maybe the first step of the bootloader will be to see if port 80 is active.

Cheers,
~Paul

closed 2018-02-01 by BillyBlaze