BillyBlaze/OctoPrint-TouchUI #183: Rotate 90º

Firesped, 2018-01-23:
I am working on a project to install an octopi with 7 inch touch screen on my printer.

I need to rotate the screen 90º. which i have done but doing the following. 

adding to /boot/config.txt to rotate the UI
`display_rotate=1`

and by using on an SSH login to rotate the touch input.
`DISPLAY=:0 xinput --set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' 0 -1 1 1 0 0 0 0 1`

I got the information for this from here https://www.raspberrypi.org/forums/viewtopic.php?t=120793
and here. https://www.raspberrypi.org/forums/viewtopic.php?t=172025

The second link was due to a change in xinputs prop names.

I'm creating a script, touchrotate.sh

#!/bin/bash
xinput --set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' 0 -1 1 1 0 0 0 0 1

I need to run this script or add that line to the autostart. Where would be the best place to do this?

Firesped, 2018-01-23:
I guess I need to read the full instructions.

closed 2018-01-23 by Firesped

BillyBlaze, 2018-01-23:
Hi Firesped,

I was just about to replying with a link, but you already found it 👍 

In case somebody lands here through google or whatsoever:
- See [the troubleshooting wiki](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#calibration-rotate-touchscreen)