BillyBlaze/OctoPrint-TouchUI #184: Close browser from keyboard?

EdGCRM, 2018-01-24:
Hi there, Not really an "issue", just my own inexperience:

I am trying to get to the command line but can't figure out how to close the TouchUI / Chromium browser using the keyboard.  I've done some Googling and have tried Alt-F4, ctrl-alt-F2, and ctrl-esc.  Not very familiar with all the tricks, so would appreciate the help.

Thanks!

BillyBlaze, 2018-01-24:
You can close TouchUI (with the autostart scripts) from the commandline with with the following commands:
- Stop  
  `sudo service touchui stop`
- Restart  
  `sudo service touchui restart`
- Start   
  `sudo service touchui start`

EdGCRM, 2018-01-24:
Apologies for probably missing some keywords, I'm new.

I am trying to get TO the command line after the TouchUI and full-screen OctoPrint has already started.  Or, how do I interrupt the boot so that it doesn't go straight to the browser?  Either option will work.  Just thinking if I need to make changes to the system.

Thanks!

BillyBlaze, 2018-01-24:
Running the commands above through SSH will close chrome and will show the commandline on the touchscreen display. There is no other way to close it otherwise.

If you want to stop touchui from starting during boot then you can edit `/etc/default/touchui` and change `START=yes` into `START=no`

EdGCRM, 2018-01-25:
Ok, good to know.  In case anyone runs into similar issues - I had set up the OctoPi server on a different network than I intended to run it, thinking I would just adjust the network stuff when I got home.  But I had no way to access it (as far as I knew) unless it was on the network (so I could do SSH).  The fix was to bring it back to the work network the next day and use SSH to add an additional network to it.

Thanks for the help - Might be fun to add an option to return to the command line (maybe with a PIN or something).  Just a thought.

-Ed

closed 2018-01-25 by EdGCRM

BillyBlaze, 2018-01-25:
Your welcome! 

Also in addition; You can add the TouchUI commands as a system commands and close it from there. See more information about that here: https://github.com/foosel/OctoPrint/wiki/Cookbook:-System-Commands