BillyBlaze/OctoPrint-TouchUI #187: Only see black screen + mouse icon

dudljump, 2018-01-31:
Hi,
I've set all up on a Raspberry Pi 3 and a Waveshare 3.5 LCD Touch Screen as it is described in the instructions. verything works, but not the touch UI itself...
I only see the mouse icon on the Touchscreen moving around when I touch on it. But that is it.. The rest of the screen is black...
If I open Octoprint on a laptop using Chrome the Touch UI works fine...
Do you have any further idea?

dudljump, 2018-01-31:
May as a further thing... 
If I restart the touch UI via ssh - it works!!! 
But as the pi starts to boot it ends up in the described situation

BillyBlaze, 2018-02-01:
Hi @dudljump,

Wow, That is a bit weird. 
Can you provide me your log located under `/var/log/xorg.0.log` after you did a reboot? (I need to read this log after Chromium failed to render)

dudljump, 2018-02-02:
HI @BillyBlaze ,
thank you for your answer. Yes that's really weird..

Here is my log-entry:
[   111.779]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[   111.780] X Protocol Version 11, Revision 0
[   111.780] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[   111.780] Current Operating System: Linux octopi 4.9.78-v7+ #1084 SMP Thu Jan 25 18:05:49 GMT 2018 armv7l
[   111.780] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[   111.780] Build Date: 11 November 2016  11:59:59AM
[   111.780] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[   111.780] Current version of pixman: 0.33.3
[   111.780]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   111.780] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   111.783] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb  1 19:14:33 2018
[   111.784] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   111.785] (==) No Layout section.  Using the first Screen section.
[   111.785] (==) No screen section available. Using defaults.
[   111.785] (**) |-->Screen "Default Screen Section" (0)
[   111.785] (**) |   |-->Monitor "<default monitor>"
[   111.786] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[   111.786] (**) |   |-->Device "touchscreen"
[   111.786] (**) |   |-->GPUDevice "Allwinner A10/A13 FBDEV"
[   111.786] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   111.786] (==) Automatically adding devices
[   111.786] (==) Automatically enabling devices
[   111.786] (==) Automatically adding GPU devices
[   111.786] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   111.786] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   111.787]    Entry deleted from font path.
[   111.787] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   111.787] (==) ModulePath set to "/usr/lib/xorg/modules"
[   111.787] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[   111.787] (II) Loader magic: 0x603f20
[   111.787] (II) Module ABI versions:
[   111.787]    X.Org ANSI C Emulation: 0.4
[   111.787]    X.Org Video Driver: 20.0
[   111.788]    X.Org XInput driver : 22.1
[   111.788]    X.Org Server Extension : 9.0
[   111.791] (--) using VT number 2

[   111.791] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   111.792] (II) no primary bus or device found
[   111.792] (II) LoadModule: "glx"
[   111.793] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   111.800] (II) Module glx: vendor="X.Org Foundation"
[   111.800]    compiled for 1.18.4, module version = 1.0.0
[   111.800]    ABI class: X.Org Server Extension, version 9.0
[   111.800] (==) AIGLX enabled
[   111.801] (II) LoadModule: "fbdev"
[... 63 lines cut ...]
[   112.049]    compiled for 1.18.4, module version = 2.10.3
[   112.049]    Module class: X.Org XInput Driver
[   112.050]    ABI class: X.Org XInput driver, version 22.1
[   112.050] (II) Using input driver 'evdev' for 'ADS7846 Touchscreen'
[   112.050] (**) ADS7846 Touchscreen: always reports core events
[   112.050] (**) evdev: ADS7846 Touchscreen: Device: "/dev/input/event0"
[   112.050] (--) evdev: ADS7846 Touchscreen: Vendor 0 Product 0
[   112.050] (--) evdev: ADS7846 Touchscreen: Found absolute axes
[   112.050] (--) evdev: ADS7846 Touchscreen: Found x and y absolute axes
[   112.050] (--) evdev: ADS7846 Touchscreen: Found absolute touchscreen
[   112.050] (**) Option "SwapAxes" "1"
[   112.050] (II) evdev: ADS7846 Touchscreen: Configuring as touchscreen
[   112.050] (**) evdev: ADS7846 Touchscreen: YAxisMapping: buttons 4 and 5
[   112.050] (**) evdev: ADS7846 Touchscreen: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[   112.050] (**) Option "config_info" "udev:/sys/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input0/event0"
[   112.051] (II) XINPUT: Adding extended input device "ADS7846 Touchscreen" (type: TOUCHSCREEN, id 6)
[   112.051] (II) evdev: ADS7846 Touchscreen: initialized for absolute axes.
[   112.051] (**) ADS7846 Touchscreen: (accel) keeping acceleration scheme 1
[   112.051] (**) ADS7846 Touchscreen: (accel) acceleration profile 0
[   112.051] (**) ADS7846 Touchscreen: (accel) acceleration factor: 2.000
[   112.051] (**) ADS7846 Touchscreen: (accel) acceleration threshold: 4
[   112.052] (II) config/udev: Adding input device ADS7846 Touchscreen (/dev/input/mouse0)
[   112.052] (**) ADS7846 Touchscreen: Applying InputClass "calibration"
[   112.052] (II) No input driver specified, ignoring this device.
[   112.052] (II) This device may have been added with another device file.

dudljump, 2018-02-07:
@BillyBlaze
Is there no solution to it?

BillyBlaze, 2018-02-07:
I don't know whats wrong yet, this log looks fine. Chromium seems running.

What OctoPi do you have installed? and what Chromium version do you have installed?

You can check the latter one with `dpkg -s chromium-browser`

BillyBlaze, 2018-11-11:
I just stumbled upon this issue myself, and I think I figured how to fix this issue.

BillyBlaze, 2018-11-11:
Temp workaround: run `sudo raspi-config` > 'Advanced options' > 'GL driver' > choose 'G1 GL (Full KMS)' hit enter and reboot

closed 2018-11-18 by BillyBlaze