BillyBlaze/OctoPrint-TouchUI #189: TouchUI screen stuck at trying to connect

Littlegiant2, 2018-02-07:
I just installed the TOUCHUI program to my Raspberry PI3 running Octopi.  Octopi had been running up to this point but the line commands were not that easy and I preffered a desktop touchscreen environment.  Since I cannot get my Windows 10 Lenovo P50 to load Octoprint (it stops loading at the plugin manager) I cannot see the PI3, nor edit the files on the 32G SD card.  I am using the Raspberry PI 7" screen.  I am very new to the Raspberry PI world so be kind.  What do I need to log, and how, in order to tell what is going on with the PI and even my Octoprint failing to start on the PC.  Thanks for any help that is offered.  Newbie here wanting to learn.

MDM63, 2018-02-27:
I have same issue. If I modify the chromium.xinit file and replace the line no:57 $(dirname "$0")/load-screen/startup.html" to $http://localhost:5000" it will connect to OctoPrint and TouchUI. So it seems, for some reason, the java script(script.js) is not working. The animation is playing (the text is flashing brighter and darker as it should), but it won't connect or time out with error or anything.

I have adapted [these](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)) instructions when installing. I had to change chromium-browser to chromium in couple places, but other than that, all should be quite the same.
I am running it on Armbian 4.14.22-sunxi. Browser is Chromium 63.0.3239.84 built on Debian 9.3, running on Debian 9.3. On OrangePi Plus, with Waveshare 7" Display.

BillyBlaze, 2018-03-14:
Sorry for the delayed response.

Thanks both for reporting this issue, I'll fix this in the upcoming release.

BillyBlaze, 2018-09-02:
Thanks all for the feedback. Could anybody perhaps install the maintenance branch to see if this issue still persists with the latest fixes?

You can uninstall TouchUI and install it with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

TheBlackMini, 2018-09-08:
Confirmed that the maintenance version works.

closed 2018-09-29 by stale[bot]