BillyBlaze/OctoPrint-TouchUI #191: [HELP!] TouchIU always "freezing" on local

bolsoncerrado, 2018-02-15:
Please HELP!

I've been a huge fan and user of Octoprint/TouchUI since the very beginning and from the last few octoprint updates, I'm experiencing TouchUI crashes on LOCAL display touch (official Raspi 7") while the raspi operates fine on the LAN via computer web browsers and such...

I've done apt-get update/upgrades to no avail and reinstalled TouchUI from scratch just in case, but as soon as it reboots and I start clicking here and there (touching the screen, that is), the thing starts to LAAAAG heavily until collapsing with a chromium "kill this app or wait" screen :/

I dont know what else to try, can't understand whats the issue.

Running on Raspi3 1.2 with official power supply.

Thanks!

bolsoncerrado, 2018-02-16:
OK I started by disabling ALL plugins except Gina's and TouchUI and it's stable now, will start investigate the plugin conflict.... I may suspect it has to do with the TAB ORDER plugin.

BillyBlaze, 2018-03-02:
Sorry for the long delay and thanks for the feedback.

Please let me know if you have found the culprit!

bolsoncerrado, 2018-03-02:
It's indeed a conflict of plugins. Not sure 109% which one but I still think it may be the Tabs one.

closed 2018-03-23 by stale[bot]

TheNore, 2018-09-16:
This is also happening to me! and I have the Tabs order plugin, unfortunately I do need to have this plugin working. I also am running version  0.3.12 of touch UI and 0.5.0 for TabsOrder