BillyBlaze/OctoPrint-TouchUI #192: iPhone x room for one more tab on top of screen?

LazaroFilm, 2018-02-16:
Hi, 
On the iPhone X, it feels like there would be enough room to display one more tab instead of having it hidden in the drop down menu. Maybe shrinking the width of the tabs just a bit would allow for that? 
Also is there a way to organize which tab is showing on the top bar? 
See screenshot: https://i.imgur.com/zGgMFcf.png

BillyBlaze, 2018-03-02:
Sorry for the long delay and thanks for the feedback.

I'll optimize the width on smaller resolutions in the upcoming release.

LazaroFilm, 2018-03-08:
Thanks! iPhone X's definition is 2436-by-1125-pixel. but the web pages are dividing the pixel count by a factor 3 so the effective screen resolution is 812px x 375px. so we are just 17 pixels shorts of displaying 4 tabs (3 icons + menu) each tab would have to be a max of 93px to fit in the screen instead of 98px

closed 2018-09-15 by BillyBlaze

BillyBlaze, 2018-09-15:
Thanks, this has been released in 0.3.12

commit bbc3f0f3 (2018-09-02): :art: Move sidebar(s) to a tab

Items in the sidebar was not displayed, in this commit items from
the sidebar will be copied to a new tab and creates a link in the
navbar and tabbar

Fixes #196

:art: Show more menu items on lower res

Fixes #192