BillyBlaze/OctoPrint-TouchUI #194: TouchUI Connection Failed

scarter311, 2018-02-27:
Hello, I am unable to get TouchUI to connect, I believe there may have been conflicting directory or driver inputs during the install but I am not familiar enough with this system to track it down I would greatly appreciate your help. I followed the LCD screen's information to install the driver then the Boot to Browser tutorial and could not get step 5 to complete, "No Such File or Dir". Using the 5" LCD driver obtained through: 

git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/

Thank you very much

scarter311, 2018-03-13:
I have still not been able to resolve this issue and would greatly appreciate some advice if anyone may be able to help, thanks

BillyBlaze, 2018-03-13:
Hi @scarter311,

Sorry for the delayed response, I barely had time to check github the last few months.
Do you use OctoPi?

scarter311, 2018-03-13:
Hello thank you for following up, yes I am using octoprint on the raspberry Pi, everything is working well except the TouchUI connection. 

Thanks, 

Steve

BillyBlaze, 2018-03-13:
There is a difference between _OctoPi_ and _OctoPrint_.

- OctoPi is a pre-configured Linux image with OctoPrint pre-installed.
- OctoPrint is printing software that can be installed on any OS.

I am asking this because to make sure you're following the right Wiki.

scarter311, 2018-03-13:
My apologies I thought they were two terms for the same system, I installed the Octoprint image and made sure that was working properly, then installed the driver for the LCD and then installed the Raspberry Pi desktop. I probably still have the exact instructions I followed on my computer if that would help.

scarter311, 2018-03-13:
I believe I installed OctoPi not Octoprint

BillyBlaze, 2018-03-14:
The installation scripts also use the [same directory](https://github.com/goodtft/LCD-show/blob/master/LCD5-show#L5).

Are you sure you didn't made a typo? or didn't select the whole command when copy-pasting?

You should be able to create a new file with `nano` once the directory is there.

scarter311, 2018-03-15:
I retried step 5 to register the screen to x11 and did not receive the error after reentering the nano input and the section, identifier, etc and now the screen is not attempting to connect to TouchUI, the system is waiting at the login screen.

Thanks

BillyBlaze, 2018-03-15:
Okay, thats what I was expecting. Could you share the contents of `/var/log/xorg.0.log`

scarter311, 2018-03-15:
Of course I still post it this afternoon. Thank you very much

scarter311, 2018-03-17:
I apologize for the delayed response but I've tried putting in that command and am getting a "no such directory" error.

BillyBlaze, 2018-03-31:
`sudo cat /var/log/Xorg.0.log` should work, if it still doesn't exist then are you sure you xserver was correctly installed?

scarter311, 2018-03-31:
pi@flashforge:~ $ sudo cat /var/log/Xorg.0.log
[    14.047]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    14.047] X Protocol Version 11, Revision 0
[    14.047] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    14.047] Current Operating System: Linux flashforge 4.14.22-v7+ #1096 SMP Mo                                                                                                             n Feb 26 19:14:22 GMT 2018 armv7l
[    14.047] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=576 bcm2708                                                                                                             _fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=                                                                                                             0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmc                                                                                                             blk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFon                                                                                                             t6x11 logo.nologo quiet splash plymouth.ignore-serial-consoles
[    14.047] Build Date: 11 November 2016  11:59:59AM
[    14.047] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    14.048] Current version of pixman: 0.33.3
[    14.048]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    14.048] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.048] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Mar 31 18:27:26 201                                                                                                             8
[    14.048] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    14.049] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.050] (==) No Layout section.  Using the first Screen section.
[    14.050] (==) No screen section available. Using defaults.
[    14.050] (**) |-->Screen "Default Screen Section" (0)
[    14.050] (**) |   |-->Monitor "<default monitor>"
[    14.051] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    14.051] (**) |   |-->Device "touchscreen"
[    14.051] (**) |   |-->GPUDevice "Allwinner A10/A13 FBDEV"
[    14.051] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    14.051] (==) Automatically adding devices
[    14.051] (==) Automatically enabling devices
[    14.051] (==) Automatically adding GPU devices
[    14.051] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.051] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.051]    Entry deleted from font path.
[    14.051] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    14.051]    Entry deleted from font path.
[    14.051] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    14.051]    Entry deleted from font path.
[    14.051] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    14.051]    Entry deleted from font path.
[    14.051] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    14.051]    Entry deleted from font path.
[    14.051] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    14.051]    Entry deleted from font path.
[    14.052] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    14.052] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.052] (II) The server relies on udev to provide the list of input devices                                                                                                             .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                             es.
[    14.052] (II) Loader magic: 0x6d5f20
[    14.052] (II) Module ABI versions:
[    14.052]    X.Org ANSI C Emulation: 0.4
[    14.052]    X.Org Video Driver: 20.0
[    14.052]    X.Org XInput driver : 22.1
[    14.052]    X.Org Server Extension : 9.0
[    14.055] (++) using VT number 7

[    14.055] (II) systemd-logind: logind integration requires -keeptty and -keep                                                                                                             tty was not provided, disabling logind integration
[    14.055] (II) no primary bus or device found
[    14.055] (II) LoadModule: "glx"
[    14.056] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    14.061] (II) Module glx: vendor="X.Org Foundation"
[    14.061]    compiled for 1.18.4, module version = 1.0.0
[    14.061]    ABI class: X.Org Server Extension, version 9.0
[    14.061] (==) AIGLX enabled
[    14.061] (II) LoadModule: "fbdev"
[    14.061] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    14.062] (II) Module fbdev: vendor="X.Org Foundation"
[... 18 lines cut ...]
[    14.065] (EE) open /dev/fb1: No such file or directory
[    14.065] (WW) Falling back to old probe method for fbdev
[    14.065] (II) Loading sub module "fbdevhw"
[    14.065] (II) LoadModule: "fbdevhw"
[... cut ...]
pi@flashforge:~ $

I apologize for the confusion I do really appreciate you not yet giving up on this issue. I have been trying the boot to commands and have had no luck hopefully I have not complicated the issue. If you think it would be best to wipe the memory card and restart I have no problem with that if you can recommend an order of operations for installing the OctoPrint software as well as the screen and drivers. 

Thank you so much, do you accept donations for your efforts in supporting issues?

BillyBlaze, 2018-03-31:
You've a weird issue, it seems like xserver found your display but failed to access it later on in the process. There might been an issue with the SPI or the kernel, therefor a new fresh install might not be a bad idea. Otherwise you might need upgrade the kernel/bootloader.

scarter311, 2018-03-31:
What order should I be installing systems. Do I need to install the raspberry pi desktop at any point, when should I install the drivers for the screen from the manufacturer?

Thank you very much

BillyBlaze, 2018-04-01:
Your welcome!

I would recommend these steps:
`Install OctoPi -> Setup LCD screen* -> Setup OctoPrint -> Install TouchUI within OctoPrint -> Setup TouchUI (wiki steps)`

`* Install the drivers, after installation you should sees no black screens or white screen but the terminal`

Also no need to install desktop, the steps from the wiki are meant to only boot the Chromium browser without any unnecessary software being run.

scarter311, 2018-04-03:
I have reformatted the SD card and reinstalled everything in the recommended order, how do I get the pi to login to the Octoprint interface, I am only seeing the text feed as I would in Putty and it waits at the login point.

BillyBlaze, 2018-04-03:
Okay, could you share `sudo cat /var/log/Xorg.0.log` again?

scarter311, 2018-04-03:
pi@Printers:~ $ sudo cat /var/log/Xorg.0.log
[     7.848]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[     7.849] X Protocol Version 11, Revision 0
[     7.849] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[     7.849] Current Operating System: Linux Printers 4.14.31-v7+ #1104 SMP Thu                                                                                                              Mar 29 16:52:18 BST 2018 armv7l
[     7.849] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708                                                                                                             _fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=                                                                                                             0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmc                                                                                                             blk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFon                                                                                                             t6x11 logo.nologo
[     7.849] Build Date: 11 November 2016  11:59:59AM
[     7.849] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[     7.849] Current version of pixman: 0.33.3
[     7.849]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     7.849] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.849] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Apr  3 09:35:27 201                                                                                                             8
[     7.903] (==) Using config directory: "/etc/X11/xorg.conf.d"
[     7.903] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.934] (==) No Layout section.  Using the first Screen section.
[     7.934] (==) No screen section available. Using defaults.
[     7.934] (**) |-->Screen "Default Screen Section" (0)
[     7.934] (**) |   |-->Monitor "<default monitor>"
[     7.947] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[     7.947] (**) |   |-->Device "touchscreen"
[     7.947] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[     7.947] (==) Automatically adding devices
[     7.947] (==) Automatically enabling devices
[     7.947] (==) Automatically adding GPU devices
[     7.957] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     7.964] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     7.964]    Entry deleted from font path.
[     7.964] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     7.964]    Entry deleted from font path.
[     7.965] (==) FontPath set to:
        built-ins
[     7.965] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.965] (II) The server relies on udev to provide the list of input devices                                                                                                             .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                             es.
[     7.965] (II) Loader magic: 0x6bcf20
[     7.965] (II) Module ABI versions:
[     7.965]    X.Org ANSI C Emulation: 0.4
[     7.965]    X.Org Video Driver: 20.0
[     7.965]    X.Org XInput driver : 22.1
[     7.965]    X.Org Server Extension : 9.0
[     7.967] (--) using VT number 2

[     7.967] (II) systemd-logind: logind integration requires -keeptty and -keep                                                                                                             tty was not provided, disabling logind integration
[     7.967] (II) no primary bus or device found
[     7.972] (II) LoadModule: "glx"
[     7.993] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     8.067] (II) Module glx: vendor="X.Org Foundation"
[     8.067]    compiled for 1.18.4, module version = 1.0.0
[     8.067]    ABI class: X.Org Server Extension, version 9.0
[     8.067] (==) AIGLX enabled
[     8.067] (II) LoadModule: "fbdev"
[     8.067] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[     8.071] (II) Module fbdev: vendor="X.Org Foundation"
[     8.071]    compiled for 1.18.4, module version = 0.4.4
[     8.071]    Module class: X.Org Video Driver
[     8.071]    ABI class: X.Org Video Driver, version 20.0
[     8.071] (II) FBDEV: driver for framebuffer: fbdev
[... cut ...]
[     8.078] (EE) No devices detected.
[     8.078] (EE)
Fatal server error:
[     8.078] (EE) no screens found(EE)
[     8.079] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[     8.079] (EE) Please also check the log file at "/var/log/Xorg.0.log" for ad                                                                                                             ditional information.
[     8.079] (EE)
[     8.087] (EE) Server terminated with error (1). Closing log file.

closed 2018-04-24 by stale[bot]