BillyBlaze/OctoPrint-TouchUI #195: Webcam Stream Not loaded

cmcdermi, 2018-03-10:
I have set up octopi and used my 3D printer with a pi cam flawlessly for a few weeks now. Yesterday I set up my raspberry pi with a 7" touch screen and successfully configured the touch UI plugin and auto start on the pi. Now I cannot seem to get my webcam to load at all. Initially it would load somewhat sluggish on my PC now it doesnt load on the desktop interface or the touch interface. I am new to this and am trying to learn my way around linux, any help is very much appreciated! 

Thank you,
-Chris

cmcdermi, 2018-03-13:
Problem Solved!
After looking at the webcam log I noticed the log saying "No Camera Detected.... Trying again in 2 minutes". Somehow the Pi camera became disabled in the raspberry pi configuration. I had been using the webcam for weeks without issue but when I installed the touch screen it somehow became disabled. I'm not sure if this was caused by the touch screen driver or the touch ui plugin, but simply using raspiconf to re-enable the pi cam solved the problem. After this the feed still did not display locally on the touch screen, however after browsing this forum I found the solution was to give an explicit address in the webcam settings. Where it says "Stream URL:" I used the following URL: "http://10.1.1.111/webcam/?action=stream" (the IP address of the PI). 

Sorry if I wasted anyone's time but hopefully this will help anyone else who may run into this problem. 

Thank you, 
-Chris

closed 2018-03-13 by cmcdermi

sengemann, 2019-06-07:
Good find! I would not have thought to look there.

Levanterman, 2020-06-03:
> Problem Solved!
> After looking at the webcam log I noticed the log saying "No Camera Detected.... Trying again in 2 minutes". Somehow the Pi camera became disabled in the raspberry pi configuration. [...]

Awesome, thanks for sharing, this fixed it for me after installing the touch screen :-)