BillyBlaze/OctoPrint-TouchUI #196: [question] Troubleshooting methodology for TouchUI incompatibility w/other plugins?

ghandye, 2018-03-11:
Hi, looking for some general guidance before submitting a formal bug report in this repo -

I use the Filament Manager plugin extensively with Octoprint. Would like to move to TouchUI for control via smartphone, but it doesn't seem to play nice with Filament Manager (the plugin's extra bits of UI do not show up in TouchUI).

There's an issue open for this problem over on the Filament Manager repo:

    https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/35

But as is common for these sorts of incompatibilities, the author of that plugin doesn't think it's their issue and has no interest in tracking it down.

Could you offer some recommendations on how to gather the necessary information to troubleshoot this and determine whether this is (or is not) a problem with TouchUI? There's no errors or warnings in the javascript console.

Thanks.

ghandye, 2018-03-16:
Thanks for the reply! Would it be helpful to file an issue (with examples) to track this, or is it already on the radar?

BillyBlaze, 2018-03-16:
Your welcome and nothing was on my radar, I could give it a look if it's easy to support in the upcoming release. 
Also no need to fill in a new issue. I think I have enough information, and if I do need some I'll post it in here.

shoreliner, 2018-03-31:
I would also love this functionality -- when working on the octoprint-klipper plugin, I'd like to be able to ensure the users TouchUI settings translate without any additional configuration, as in many cases their LCD screen may not even be functional(Firmware still not there yet for most displays).

jneilliii, 2018-07-21:
I've also been asked to make my new [OctoPrint-BedLevelingWizard](https://github.com/jneilliii/OctoPrint-BedLevelingWizard/issues/5) plugin TouchUI compatible.  If I can help in any way in moving this development forward please let me know. I'm going to start investigating if there are any potential hacks to allow my viewmodel to load in TouchUI.  If I were to dynamically move the sidebar viewmodel to a tab panel when TouchUI is enabled, would that allow TouchUI to load it within it's interface?

jneilliii, 2018-07-21:
I was able to accomplish functionality using the following in my view model's js file.  Not sure if this will help others in adding compatibility to their own plugins, but hope it helps someone.

```
self.onStartupComplete = function () {
	// move elements if using touchui
	if($('html[id="touch"]').length > 0){
		$('#touchui_dropdown_link').before('<li id="sidebar_plugin_<put your plugin id here>_link" class=""><a href="#sidebar_plugin_<put your plugin id here>" data-toggle="tab"></a></li>');
		$('#tabs_content').append($('#sidebar_plugin_<put your plugin id here>').addClass('tab-pane'));
	}
}
```

jneilliii, 2018-07-21:
So I've come to realize this works on a larger screen but whatever process puts the tabs into the menu in the upper right corner doesn't recognize this new link being appended and therefore is created overlapping the existing menu button.  Any advice would be greatly appreciated.

jneilliii, 2018-07-22:
Ok, after reverse engineering TouchUI I now have a functional hack for plugin developers for a sidebar panel.  It's working for me with the OctoPrint-BedLevelingWizard plugin.  Your mileage may vary.

```
self.onAfterBinding = function () {
	// touchui hack
	if($('html[id="touch"]').length > 0){
		$('#touchui_dropdown_link').before('<li id="sidebar_plugin_<put your plugin id here>_wrapper_link" class=""><a href="#sidebar_plugin_<put your plugin id here>" data-toggle="tab"></a></li>');
		$('#tabs_content').append($('#sidebar_plugin_<put your plugin id here>').addClass('tab-pane'));
		
		var $elm = $('#sidebar_plugin_<put your plugin id here>_wrapper_link')			
		$elm
			.clone()
			.attr("id", $elm.attr("id")+"2")
			.prependTo("#all_touchui_settings > .dropdown-menu")
			.find("a")
			.text('<Put link text to show in side menu here>')
			.off("click")
			.on("click", function(e) {
				$elm.find('a').click();
				$("#all_touchui_settings").addClass("item_active");
				e.preventDefault();
				return false;
			});
		$elm.addClass('hidden_touch');
		$(window).trigger('resize');
	}
}
```

[image]

commit bbc3f0f3 (2018-09-02): :art: Move sidebar(s) to a tab

Items in the sidebar was not displayed, in this commit items from
the sidebar will be copied to a new tab and creates a link in the
navbar and tabbar

Fixes #196

:art: Show more menu items on lower res

Fixes #192
closed 2018-09-15 by BillyBlaze