BillyBlaze/OctoPrint-TouchUI #200: Kioks install missing Touch UI Configuration

ianjolin, 2018-03-21:
Ive installed the kiosk and browser required but there is no default configuration for Octopi TouchUI. While I was walking through the steps the move of the default TouchUI files might have error-ed out.

I can launch the minimum Kioks interface but I am unsure how to configure kiosk to move forward.

BillyBlaze, 2018-03-21:
Hi @ianjolin 

Support for the kiosk browser has been stopped two years ago, I have no idea if the code made for "Wheezy"  still work on "Jessie".

Anyhow, the only method currently supported is with Chromium, [See the wiki here](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)).

I suggest to remove the folder with legacy code and start again from step 1 in above link
```
rm -rf ~/TouchUI-autostart/
```

ianjolin, 2018-03-22:
I followed the instruction up to the reboot and received the following. As you cans see via my VNC connection the Kiosk doesn’t seem to launch. I had received errors as defined in step 6 and 9 (Second bullet point). I ma curious if I should be doing something more.

ianjolin, 2018-03-22:
Which AutoStart file should be configured to define the url? This one has the following:

BillyBlaze, 2018-03-23:
@ianjolin, please use github.com when posting images.

closed 2018-04-13 by stale[bot]