BillyBlaze/OctoPrint-TouchUI #202: Booting to browser from one pi to to another with OctoPrint.

pokemonw2, 2018-03-28:
I have set up the boot to browser script according to the wiki but would like it to boot to another pi with OctoPrint instead of running of it locally. I have tried to look search around in the script files and in the wiki but have not found a way to change the IP address which the script boots to.

Does anyone know how to change this?

BillyBlaze, 2018-03-28:
Hi @pokemonw2,

You can change the variable `prefix` in  `~/TouchUI-autostart/load-screen/script.js` from `localhost` to the ip address or hostname.

That should fix it for now.

In the future I will make this a bit more easily configurable.

BillyBlaze, 2018-03-29:
Just checking: You have installed TouchUI within OctoPrint on 192.168.1.107 ?

If you did, then make sure you can connect to OctoPrint with the same url with your browser, if that works, then just replace the whole string with it.

pokemonw2, 2018-03-30:
Thanks! That did the trick.

closed 2018-03-30 by pokemonw2

BillyBlaze, 2018-03-30:
Awesome and your welcome!

Could you perhaps share your solution for in case somebody else wonders into this thread?

k-eye, 2020-03-28:
Hi there.

Im trying to do the same like pokemon, but touchui will not connect because it dont has the permission because of missing username and pw from the octoprint on the other pi.
Can anybody please tell me where to enter this information? ive only found where to enter it, if there is octoprint installed, what in my case isnt on the „remote“-pi...
thx