BillyBlaze/OctoPrint-TouchUI #203: octopi raspbery 3 touchscreen officilal 7" probleme

shinji99, 2018-03-28:
hello 

iam new and noob , 
i have a raspberry pi 3 with offical 7" touchscreen (nappe DSI) with official octopi 0.14.0 distribution octoprint 1.3.6 . Touch ui is V 0.3.11 version
i use this tutorial
 https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

but i can't boot to touch ui i boot many time to temrinal i don't know what to do 

thanks in advance and sorry for my english

shinji99, 2018-03-29:
my log:

[code]pi@octopi:~ $ xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43                                                                                                                                                       BST 2017 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=                                                                                                                                                      480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  d                                                                                                                                                      wc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=ba5a79eb-02                                                                                                                                                       rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 11:08:16 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional info                                                                                                                                                      rmation.
(EE)
(EE) Server terminated with error (1). Closing log file.
[code]

shinji99, 2018-03-29:
i have try the 2 value for:
Section "Device"  
  Identifier "myfb"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection

fb0 and fb1

shinji99, 2018-03-29:
other log files:
pi@octopi:~ $ sudo cat /var/log/Xorg.0.log
[    50.508]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    50.508] X Protocol Version 11, Revision 0
[    50.508] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    50.508] Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l
[    50.508] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=ba5a79eb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    50.508] Build Date: 11 November 2016  11:59:59AM
[    50.508] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    50.509] Current version of pixman: 0.33.3
[    50.509]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    50.509] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    50.509] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 11:08:16 2018
[    50.510] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    50.511] (==) No Layout section.  Using the first Screen section.
[    50.511] (==) No screen section available. Using defaults.
[    50.511] (**) |-->Screen "Default Screen Section" (0)
[    50.511] (**) |   |-->Monitor "<default monitor>"
[    50.512] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    50.512] (**) |   |-->Device "touchscreen"
[    50.512] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    50.512] (==) Automatically adding devices
[    50.512] (==) Automatically enabling devices
[    50.512] (==) Automatically adding GPU devices
[    50.512] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    50.512] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    50.512]    Entry deleted from font path.
[    50.512] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    50.512]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (==) FontPath set to:
        built-ins
[    50.513] (==) ModulePath set to "/usr/lib/xorg/modules"
[    50.513] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    50.513] (II) Loader magic: 0x54cf0f20
[    50.513] (II) Module ABI versions:
[    50.513]    X.Org ANSI C Emulation: 0.4
[    50.513]    X.Org Video Driver: 20.0
[    50.513]    X.Org XInput driver : 22.1
[    50.514]    X.Org Server Extension : 9.0
[    50.517] (--) using VT number 2

[    50.517] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    50.518] (II) no primary bus or device found
[    50.518] (II) LoadModule: "glx"
[    50.519] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    50.526] (II) Module glx: vendor="X.Org Foundation"
[    50.526]    compiled for 1.18.4, module version = 1.0.0
[    50.526]    ABI class: X.Org Server Extension, version 9.0
[    50.526] (==) AIGLX enabled
[    50.526] (II) LoadModule: "fbdev"
[    50.527] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    50.527] (II) Module fbdev: vendor="X.Org Foundation"
[    50.527]    compiled for 1.18.4, module version = 0.4.4
[    50.527]    Module class: X.Org Video Driver
[    50.527]    ABI class: X.Org Video Driver, version 20.0
[    50.527] (II) FBDEV: driver for framebuffer: fbdev
[    50.535] (WW) Falling back to old probe method for fbdev
[    50.535] (II) Loading sub module "fbdevhw"
[    50.535] (II) LoadModule: "fbdevhw"
[    50.536] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    50.536] (II) Module fbdevhw: vendor="X.Org Foundation"
[    50.536]    compiled for 1.18.4, module version = 0.0.2
[    50.536]    ABI class: X.Org Video Driver, version 20.0
[    50.537] (EE) open /dev/fb1: No such file or directory
[    50.537] (EE) No devices detected.
[    50.537] (EE)
Fatal server error:
[    50.537] (EE) no screens found(EE)
[    50.537] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    50.537] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    50.537] (EE)
[    50.552] (EE) Server terminated with error (1). Closing log file.

shinji99, 2018-03-29:
and the as log 
pi@octopi:~ $ sudo cat /var/log/Xorg.0.log
[    50.508]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    50.508] X Protocol Version 11, Revision 0
[    50.508] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    50.508] Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun                                                                                                                                                       30 14:47:43 BST 2017 armv7l
[    50.508] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708                                                                                                                                                      _fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=                                                                                                                                                      0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID                                                                                                                                                      =ba5a79eb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    50.508] Build Date: 11 November 2016  11:59:59AM
[    50.508] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    50.509] Current version of pixman: 0.33.3
[    50.509]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    50.509] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    50.509] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 11:08:16 201                                                                                                                                                      8
[    50.510] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    50.511] (==) No Layout section.  Using the first Screen section.
[    50.511] (==) No screen section available. Using defaults.
[    50.511] (**) |-->Screen "Default Screen Section" (0)
[    50.511] (**) |   |-->Monitor "<default monitor>"
[    50.512] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    50.512] (**) |   |-->Device "touchscreen"
[    50.512] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    50.512] (==) Automatically adding devices
[    50.512] (==) Automatically enabling devices
[    50.512] (==) Automatically adding GPU devices
[    50.512] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    50.512] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    50.512]    Entry deleted from font path.
[    50.512] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    50.512]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    50.513]    Entry deleted from font path.
[    50.513] (==) FontPath set to:
        built-ins
[    50.513] (==) ModulePath set to "/usr/lib/xorg/modules"
[    50.513] (II) The server relies on udev to provide the list of input devices                                                                                                                                                      .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                                                                      es.
[    50.513] (II) Loader magic: 0x54cf0f20
[    50.513] (II) Module ABI versions:
[    50.513]    X.Org ANSI C Emulation: 0.4
[    50.513]    X.Org Video Driver: 20.0
[    50.513]    X.Org XInput driver : 22.1
[    50.514]    X.Org Server Extension : 9.0
[    50.517] (--) using VT number 2

[    50.517] (II) systemd-logind: logind integration requires -keeptty and -keep                                                                                                                                                      tty was not provided, disabling logind integration
[    50.518] (II) no primary bus or device found
[    50.518] (II) LoadModule: "glx"
[    50.519] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    50.526] (II) Module glx: vendor="X.Org Foundation"
[    50.526]    compiled for 1.18.4, module version = 1.0.0
[    50.526]    ABI class: X.Org Server Extension, version 9.0
[    50.526] (==) AIGLX enabled
[    50.526] (II) LoadModule: "fbdev"
[    50.527] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    50.527] (II) Module fbdev: vendor="X.Org Foundation"
[    50.527]    compiled for 1.18.4, module version = 0.4.4
[    50.527]    Module class: X.Org Video Driver
[... cut ...]
Fatal server error:
[    50.537] (EE) no screens found(EE)
[    50.537] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    50.537] (EE) Please also check the log file at "/var/log/Xorg.0.log" for ad                                                                                                                                                      ditional information.
[    50.537] (EE)
[    50.552] (EE) Server terminated with error (1). Closing log file.

shinji99, 2018-03-29:
log with fb0:

pi@octopi:~ $ sudo cat /var/log/Xorg.0.log
[    13.498]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    13.498] X Protocol Version 11, Revision 0
[    13.498] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    13.498] Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun                                                                                                                                                       30 14:47:43 BST 2017 armv7l
[    13.498] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708                                                                                                                                                      _fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=                                                                                                                                                      0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID                                                                                                                                                      =ba5a79eb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    13.498] Build Date: 11 November 2016  11:59:59AM
[    13.498] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    13.498] Current version of pixman: 0.33.3
[    13.498]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    13.498] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    13.498] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 11:17:11 201                                                                                                                                                      8
[    13.520] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    13.535] (==) No Layout section.  Using the first Screen section.
[    13.535] (==) No screen section available. Using defaults.
[    13.535] (**) |-->Screen "Default Screen Section" (0)
[    13.535] (**) |   |-->Monitor "<default monitor>"
[    13.547] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    13.547] (**) |   |-->Device "touchscreen"
[    13.547] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    13.547] (==) Automatically adding devices
[    13.547] (==) Automatically enabling devices
[    13.547] (==) Automatically adding GPU devices
[    13.558] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    13.563] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    13.563]    Entry deleted from font path.
[    13.563] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    13.563]    Entry deleted from font path.
[    13.563] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    13.563]    Entry deleted from font path.
[    13.564] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    13.564]    Entry deleted from font path.
[    13.564] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    13.564]    Entry deleted from font path.
[    13.564] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    13.564]    Entry deleted from font path.
[    13.564] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    13.564]    Entry deleted from font path.
[    13.564] (==) FontPath set to:
        built-ins
[    13.564] (==) ModulePath set to "/usr/lib/xorg/modules"
[    13.564] (II) The server relies on udev to provide the list of input devices                                                                                                                                                      .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                                                                      es.
[    13.564] (II) Loader magic: 0x54d81f20
[    13.564] (II) Module ABI versions:
[    13.564]    X.Org ANSI C Emulation: 0.4
[    13.564]    X.Org Video Driver: 20.0
[    13.564]    X.Org XInput driver : 22.1
[    13.564]    X.Org Server Extension : 9.0
[    13.569] (--) using VT number 2

[    13.569] (II) systemd-logind: logind integration requires -keeptty and -keep                                                                                                                                                      tty was not provided, disabling logind integration
[    13.569] (II) no primary bus or device found
[    13.573] (II) LoadModule: "glx"
[    13.590] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    13.663] (II) Module glx: vendor="X.Org Foundation"
[    13.663]    compiled for 1.18.4, module version = 1.0.0
[    13.663]    ABI class: X.Org Server Extension, version 9.0
[    13.663] (==) AIGLX enabled
[    13.664] (II) LoadModule: "fbdev"
[    13.664] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[... 103 lines cut ...]
[    14.440] (**) Option "AccelerationScheme" "none"
[    14.440] (**) FT5406 memory based driver: (accel) selected scheme none/0
[... cut ...]
[    14.452] (II) XINPUT: Adding extended input device "2.4G Wireless Touchpad K                                                                                                                                                      eyboard" (type: KEYBOARD, id 9)
[    14.452] (**) Option "xkb_model" "pc105"
[    14.452] (**) Option "xkb_layout" "gb"
[    14.632] (II) UnloadModule: "libinput"
[    14.632] (II) UnloadModule: "libinput"
[    14.633] (II) UnloadModule: "libinput"
[    14.633] (II) UnloadModule: "libinput"
[    14.642] (II) Server terminated successfully (0). Closing log file.

BillyBlaze, 2018-03-29:
use `fb0` and change your raspi-config to `start to command line`. if that doesn't work then share the xorg log again,

shinji99, 2018-03-29:
i have make rapis-config 
change boot to B1 or B2 but nothing new (boot to console with or without autolog)
what have i missing?

my log with choise B2 (autlog as pi user) 

```
pi@octopi:~ $ sudo cat /var/log/Xorg.0.log
[    14.653]
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    14.654] X Protocol Version 11, Revision 0
[    14.654] Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
[    14.654] Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun                                                                                                                                                       30 14:47:43 BST 2017 armv7l
[    14.654] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708                                                                                                                                                      _fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=                                                                                                                                                      0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID                                                                                                                                                      =ba5a79eb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    14.654] Build Date: 11 November 2016  11:59:59AM
[    14.654] xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
[    14.654] Current version of pixman: 0.33.3
[    14.654]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    14.654] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.654] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 18:48:50 201                                                                                                                                                      8
[    14.681] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.693] (==) No Layout section.  Using the first Screen section.
[    14.693] (==) No screen section available. Using defaults.
[    14.693] (**) |-->Screen "Default Screen Section" (0)
[    14.693] (**) |   |-->Monitor "<default monitor>"
[    14.698] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    14.698] (**) |   |-->Device "touchscreen"
[    14.698] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    14.699] (==) Automatically adding devices
[    14.699] (==) Automatically enabling devices
[    14.699] (==) Automatically adding GPU devices
[    14.710] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.717] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    14.717]    Entry deleted from font path.
[    14.717] (==) FontPath set to:
        built-ins
[    14.717] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.717] (II) The server relies on udev to provide the list of input devices                                                                                                                                                      .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                                                                      es.
[    14.717] (II) Loader magic: 0x54d42f20
[    14.717] (II) Module ABI versions:
[    14.717]    X.Org ANSI C Emulation: 0.4
[    14.717]    X.Org Video Driver: 20.0
[    14.717]    X.Org XInput driver : 22.1
[    14.717]    X.Org Server Extension : 9.0
[    14.720] (--) using VT number 2

[    14.720] (II) systemd-logind: logind integration requires -keeptty and -keep                                                                                                                                                      tty was not provided, disabling logind integration
[    14.720] (II) no primary bus or device found
[    14.723] (II) LoadModule: "glx"
[    14.732] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    14.795] (II) Module glx: vendor="X.Org Foundation"
[... 109 lines cut ...]
[    15.411] (**) FT5406 memory based driver: (accel) selected scheme none/0
[    15.411] (**) FT5406 memory based driver: (accel) acceleration factor: 2.000
[    15.411] (**) FT5406 memory based driver: (accel) acceleration threshold: 4
[... cut ...]
[    15.425] (II) XINPUT: Adding extended input device "2.4G Wireless Touchpad K                                                                                                                                                      eyboard" (type: KEYBOARD, id 9)
[    15.425] (**) Option "xkb_model" "pc105"
[    15.425] (**) Option "xkb_layout" "gb"
[    15.573] (II) UnloadModule: "libinput"
[    15.573] (II) UnloadModule: "libinput"
[    15.573] (II) UnloadModule: "libinput"
[    15.573] (II) UnloadModule: "libinput"
[    15.582] (II) Server terminated successfully (0). Closing log file.
 ```

shinji99, 2018-03-29:
that's happened:

pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43                                                                                                                                                       BST 2017 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=                                                                                                                                                      480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  d                                                                                                                                                      wc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=ba5a79eb-02                                                                                                                                                       rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 29 19:08:31 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
sed: can't read /home/root/.config/chromium/Default/Preferences: No such file or                                                                                                                                                       directory
sed: can't read /home/root/.config/chromium/Default/Preferences: No such file or                                                                                                                                                       directory
sed: can't read /home/root/.config/chromium/Default/Preferences: No such file or                                                                                                                                                       directory
sed: can't read /home/root/.config/chromium/Default/Preferences: No such file or                                                                                                                                                       directory
TouchUI: touch events enabled
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!                                                                                                                                                        errno=2
[1:1:0329/190834.580070:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper                                                                                                                                                       startup ack (0 bytes)
[955:955:0329/190836.663466:ERROR:desktop_window_tree_host_x11.cc(1127)] Not imp                                                                                                                                                      lemented reached in virtual void views::DesktopWindowTreeHostX11::InitModalType(                                                                                                                                                      ui::ModalType)
[1001:1001:0329/190836.681832:ERROR:gl_surface_glx.cc(406)] glxQueryVersion fail                                                                                                                                                      ed
[1001:1001:0329/190836.681978:ERROR:gl_initializer_x11.cc(130)] GLSurfaceGLX::In                                                                                                                                                      itializeOneOff failed.
[1001:1001:0329/190836.708670:ERROR:gpu_child_thread.cc(327)] Exiting GPU proces                                                                                                                                                      s due to errors during initialization

(chromium-browser:955): GConf-WARNING **: Client failed to connect to the D-BUS                                                                                                                                                       daemon:
/usr/bin/dbus-launch terminated abnormally without any error message
[955:985:0329/190836.842878:ERROR:browser_gpu_channel_host_factory.cc(113)] Fail                                                                                                                                                      ed to launch GPU process.
[955:985:0329/190836.844431:ERROR:browser_gpu_channel_host_factory.cc(113)] Fail                                                                                                                                                      ed to launch GPU process.
[... cut ...]
[955:985:0329/190836.845898:ERROR:browser_gpu_channel_host_factory.cc(113)] Fail                                                                                                                                                      ed to launch GPU process.
^Cxinit: connection to X server lost

waiting for X server to shut down [955:955:0329/190907.145458:ERROR:chrome_browser_main_extra_parts_x11.cc(62)] X IO error received (X server probably went away)
(II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2

BillyBlaze, 2018-03-29:
Seems like Chrome is already running. run `sudo service touchui stop` and try again

BillyBlaze, 2018-03-29:
use boot to console without autologin

shinji99, 2018-03-29:
another question, can i make an url request with octoprint to power on off my printer ?(i use jeedom for domotique)

closed 2018-04-19 by stale[bot]