BillyBlaze/OctoPrint-TouchUI #204: Cannot calibrate the touchscreen

waityn4jc, 2018-04-05:
I have a Raspberry Pi 3 with an Adafruit 3.2" resistive touch screen.  I had multiple of these working in the past with TouchUI.  However, I had not used it in a while and needed a reformat.  I installed Octopi's current image and then followed Adafruit's easy install which is different from the last time I did this.  I then followed your installation instructions for TouchUI.  Everything seems ok except the touchscreen seems to be calibrated about 1" away from where I actually touch.  I have attached a video showing this.  I tried both your instructions on re-calibrating and the instructions on Adafruit's website.  When trying yours I get an error when trying to save the file as it says that files does not exist.  Adafruit's acts like it has re-calibrated but then nothing changes.  Can you please help?
[IMG_1008.zip](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/1881447/IMG_1008.zip)

BillyBlaze, 2018-04-08:
Hi @waityn4jc ,

The file should not exist the first time you run the calibration. Nano should be able to create the file unless the directory has not been created, has `/etc/X11/xorg.conf.d/` been created?

If the directory is missing, you could also try to add the file to `/usr/share/X11/xorg.conf.d/`

waityn4jc, 2018-04-08:
That calibration appears to still not have fixed the problem.  So, as I watch it more closely, not only does it click away from where I click, it swipes/scrolls a different direction.  If I swipe down, the mouse actually goes to the right.  I have reformatted two other machines now and I am getting the same issue.  The new Adafruit “Easy Install” now has you choose an orientation.  It’s like the display is showing the correct orientation but the touchscreen is on a different orientation.  How do you correct the orientation of the touchscreen?  I tried to read online about it, but I got quite confused.  I believe this link speaks to it, but it seems like Greek to me.  Also, I have resistive screens, not capacitive ones.

 

http://www.0xf8.org/2016/01/complete-rotation-support-for-the-adafruit-pitft-2-8-capacitive-touchscreen-display/

waityn4jc, 2018-04-08:
I use these directions from Adafruit:

https://cdn-learn.adafruit.com/downloads/pdf/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi.pdf

 

First I run this:

sudo TSLIB_FBDEVICE=/dev/fb1 TSLIB_TSDEVICE=/dev/input/touchscreen ts_calibrate

 

 

I follow the instructions on the screen to calibrate it.  Then I test it as they request with:

 

sudo TSLIB_FBDEVICE=/dev/fb1 TSLIB_TSDEVICE=/dev/input/touchscreen ts_test

 

 

The test works fine and everything on the touchscreen works perfect.  However, once I reboot and the computer boots into TouchUI the orientation of the touchscreen is wrong.  The display is correct, but tapping on the screen is wrong.

waityn4jc, 2018-04-08:
The directory /etc/X11/xorg.conf.d/ has not been created.  Therefore, I added it to /usr/share/X11/xorg.conf.d/ per your request.  However, once TouchUI loads, the x/y axis is still off.

BillyBlaze, 2018-04-08:
Did you remove the old calibration file of XServer after the ts_calibrate steps? (e.g. 99-calibration.conf)

First fix calibration on system level (e.g. ts_calibrate) then calibrate XServer if needed. Also are you still having issues with swiping? or is it just off?

waityn4jc, 2018-04-08:
I attached another video file showing the issue I still get in TouchUI.

First, let me apologize because my Linux knowledge is basically 0.  Therefore, I am unsure if I have done any of the things you are suggesting in your last comment.  Here are the steps I have taken the best I know:

Download and flash jessie lite from Octopi website.
I set up Octoprint through the web server and then download the TouchUI extension.
Run Adafruit Easy Installer choosing option 1 for 90 degree rotation and choose yes for PiTFT as Text Console.
I then follow your instructions to boot to browser here:
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)
I then have tried 3 different troubleshooting suggestions here https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting
I tried Calibrating Touch Location, Calibration Rotate Touchscreen, and Inverted X/Y axis.

[Video of swiping.zip](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/1887711/Video.of.swiping.zip)

waityn4jc, 2018-04-08:
Perhaps I don't know what my device name is then, because I thought I
already had fixed it.  Here is the info I have used:

pi@octopi:~ $ FRAMEBUFFER=/dev/fb1 & DISPLAY=:0.0 xinput list
[1] 1559
⎡ Virtual core pointer                          id=2    [master pointer
(3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer
(2)]
⎜   ↳ stmpe-ts                                  id=6    [slave  pointer
(2)]
⎜   ↳ Microsoft Microsoft Wireless Optical Mouse® 1.00  id=7    [slave
pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard
(2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard
(3)]
[1]+  Done                    FRAMEBUFFER=/dev/fb1

Therefore, this is what I had in the file:

su $TOUCHUI_USER -c "xinput set-prop '/dev/fb1' 'Coordinate Transformation
Matrix' 0 1 0 -1 0 1 0 0 1"

BillyBlaze, 2018-04-08:
Your device name is `stmpe-ts`

waityn4jc, 2018-04-09:
I have attached a picture showing what the file says. I am still getting the same error with the touchscreen. I’m sorry if there is something I’m missing, but I can’t seem to get this to work. :(

>

BillyBlaze, 2018-04-09:
Please don't email pictures to github, they do not show up. Upload pictures with the web interface.

closed 2018-04-30 by stale[bot]

Xercium, 2020-08-15:
Hello, 
I have a 5 inch touchscreen and I am running into a calibration offset issue. I tried to follow the troubleshooting guide, but I am not getting to step 4.
[image]

Am I doing something wrong? Thanks for the help.