BillyBlaze/OctoPrint-TouchUI #205: Custom Links (Navbar) not properly displayed

emphasize, 2018-04-06:
Hi Salandora,

I've created a tiny plugin that add links to my navbar (basicly a simple advancement to Ginas' plugin tutorial). In octoprint itself it's shown properly, but as i enter TouchUI it gets attached to the seetings tab and only showing the svgs' (which i added to the Link buttons) - slightly deranged - , but no link and no name.

Code ..._navbar.jinja2 / css
```
<div class="NavbarLinks"> 
   <a href="#" class="btn home" data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url1, title: settings.settings.plugins.NavbarLinks.name1}"> 
   <span class="home-button"> </span>
   <span data-bind="text: settings.settings.plugins.NavbarLinks.name1"></span>
   </a>
    
   <a href="#" class="btn cloud" data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url2, title: settings.settings.plugins.NavbarLinks.name2}">
   <span class="cloud-button"> </span>
   <span data-bind="text: settings.settings.plugins.NavbarLinks.name2"></span>
   </a> 

   <a href="#" class="btn smarthome" data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url3, title: settings.settings.plugins.NavbarLinks.name3}">
   <span class="smarthome-button"> </span>
   <span data-bind="text: settings.settings.plugins.NavbarLinks.name3"></span>
   </a> 
</div>

#navbar .navbar-inner .NavbarLinks .home-button{

float: left;
width: 25px;
height: 16px;
padding-top: 3px;
background-image: url("../img/home.svg");
background-repeat: no-repeat;
}
.
.
.
`

What do i have to change that it'll be displayed properly?
Is there some trick that it will show up in the main tab?

Thanks,
Swen

BillyBlaze, 2018-04-08:
Hi Swen,

Do you perhaps have a screenshot of your issue so I can wrap my head around your issue a bit better?

emphasize, 2018-04-08:
Sure

[image]

BillyBlaze, 2018-04-08:
Thanks, Alrighty:

1. The "?" icon is TouchUI's attempt to make the styling of the icons more consistent. However plugins that are not 'supported' get a question mark. To change the question marks or to hide it, you could target the `:before` in your CSS with `content: none !important` to hide the questionmark [1].

2. TouchUI currently loops through all `<li>` items and seeks if the first child element is a link tag, if it's not a link, then it thinks it's a textual plugin (e.g. NavbarTemp) and places it below the login/settings.

    If you would like to show it as 'tab' on top of the dropdown, then currently you need need to remove the `<div class="NavbarLinks">` and let the first child element be a link tag. 

[1]:
```
#touchui #navbar .navbar-inner .NavbarLinks .btn:before {
    content: none !important;
}
```

emphasize, 2018-04-08:
Many thanks for the input.

The css side is going well with 

```
#touch #navbar .navbar-inner .NavbarLinks .btn:before {
    content: none !important;
}

#touch #navbar .navbar-inner .NavbarLinks .btn {
    min-height: 30px;
    padding: 4px 0 0 15px;
    border: 0px;
}
```
[image]

Maybe you can shed some light how to deal with the Link problem.

(in Touchui) the Link-url is taken from the href atrribute of the a-tag. As you can see the NavbarLinks' url is fetched over the data-bind argument. (The same way name1 is fetched)

 ```
<a href="_THERE SHOULD IT BE_" class="btn home" data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url1}">
   <span class="home-button"> </span>
   <span data-bind="text: settings.settings.plugins.NavbarLinks.name1"></span>
</a>
```

since i'm pretty new to this, I'd like to ask:

1. How is the data-bind passed to the href attribute?
2. How to deal with the second span (name1)?

BillyBlaze, 2018-04-08:
It seems like knockout is not binding your viewModel correctly, or it might be a bug in TouchUI.

TouchUI moves your items before knockout applies the viewModel, so if the id of your `<li>` remains then it should be able to bind your viewModel. However, since you don't have text nor has the href been applied.

If you run this in the console of your browser after the page has been loaded, does it specify your viewModel?

```
ko.dataFor($('#your-id-of-the-li'));
```

emphasize, 2018-04-08:
ko-data is undefined (but then again: which id you were looking for?

(I rewrote it)
```
<li id="NavbarLinks">
   <a id="home-link" href="#" class="btn home" data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url1}">
   <span class="home-button"></span>
   <span class="home-text" data-bind="text: settings.settings.plugins.NavbarLinks.name1"></span>
   </a>
```

**BUT** I am not shure if i have expressed myself correctly.

If i replace 
`<a ... href="#"`
with
`<a ... href="192.168.188.2"`

and add
`title="Home"`
to
`<a id="home-link" href="#" class="btn home"`

then 

[image]

must recode css though to see the svg.

_**or should Touchui already know what to do with**_

` data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url1}` ?

emphasize, 2018-04-08:
Maybe add a little context here

If i would write the code with the same (touchui) logic

`
   <a href="#" class="btn home" title="text: settings.settings.plugins.NavbarLinks.name1" ...`
  
(besides the point that i don't know how to correctly pass the values to title :) )

it would only displayed as a "commentary" (sorry for not useing proper language)

emphasize, 2018-04-08:
ok, i believe i have answered the question myself:

`data-bind="attr: {href: settings.settings.plugins.NavbarLinks.url1, title: settings.settings.plugins.NavbarLinks.name1}" `

should do the trick already