BillyBlaze/OctoPrint-TouchUI #206: Deactivate the zoom function

TheGim, 2018-04-15:
Is there a possibility to deactivate the zoom function in TouchUI?
I want the displayed content on the touchscreen fixed and not zoomable.

BillyBlaze, 2018-04-15:
Hi @TheGim,

I already [disable zooming](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/source/js/core/_init.js#L22) with the standard `<meta name="viewbox">` method.

Could you perhaps enlighten me some more with your issue/feature request?

TheGim, 2018-04-15:
@BillyBlaze 

On my smartphone I see the zoom being disabled. I have got the problem on a touchscreen which is directly attached to the Pi which is running OctoPrint. I'm looking for a method to disable it on the touchscreen because if I put a second finger on the touchscreen by mistake, the zoom goes crazy.

BillyBlaze, 2018-04-15:
Ahh, yeah after googling around I see I understand the issue, we need to manual disable.

Solution:
Edit `/etc/default/touchui` and add  `TOUCHUI_ARGS="--no-touch-pinch"` to the file. (it doesn't matter where you add it)

I am unable to have another solution for this issue sadly. It would've been great if Chromium just behaved towards what the <meta> specified.

TheGim, 2018-04-15:
I tried it, but it didn't work.
Via SSH, I used "sudo nano /etc/default/touchui" and added TOUCHUI_ARGS="--no-touch-pinch" but no changes so far.

I added a gif to show the problem.
[image]

BillyBlaze, 2018-04-15:
What if you try `--disable-pinch`?

TheGim, 2018-04-15:
Perfect, thank you very much, now it works flawlessly.

And I really appreciate your projekt, thank you for this too.

BillyBlaze, 2018-04-15:
Awesome! and your welcome ofc!

closed 2018-04-16 by TheGim