BillyBlaze/OctoPrint-TouchUI #207: Webcam works, but "webcam stream not loaded" message shows up

salfter, 2018-04-16:
I was playing around with Touch UI on an old tablet that I might dedicate to printer control...looks like it's mostly working, but the webcam view looks like the attached screenshot:

[image]

(If it matters, the device is a Samsung Galaxy Tab 2 7.0 (did I mention it was old?), running Lineage OS 13.  I've rooted it and installed Chrome, and I launch OctoPrint from a homescreen shortcut.)

salfter, 2018-04-16:
Another datapoint: the same thing happens with a Moto Z Play running the latest available Android release.

salfter, 2018-04-16:
Another datapoint: I'm running the latest OctoPrint, installed onto Raspbian.

BillyBlaze, 2018-05-08:
Sorry for the long delay, I am unable to look at this and will be looking into this as soon as I am able to.

Cub3squar3d, 2018-05-11:
The issue might be related to the interaction with another plugin: WebcamTab.
Since TouchUi already moves che webcam to it's own tab.
Uninstalling the plugin resolves the issue.

BillyBlaze, 2018-05-11:
Thanks for the feedback @Cub3squar3d, seems like both plugins try to do the same.

@salfter can you confirm this is the same issue for you?

If yes, then we can fix this by disabling WebcamTab if TouchUI is active

salfter, 2018-05-15:
I just got my OctoPrint setup fully running again a couple of hours ago (needed a longer cable for the camera).  My printer's  busy with a print job right now; I'll try disabling WebcamTab when it's done.

salfter, 2018-05-15:
Looks like disabling WebcamTab did the trick.

BillyBlaze, 2018-09-01:
Thanks for the confirmation, in the upcoming release the webcamTab plugin will be disabled when TouchUI is active.

@AfroPsycho, if you have a different issue then the original poster then please create a new issue, I cannot keep focus if multiple issues are reported in one issue.

AfroPsycho, 2018-09-02:
If upcoming release won't solve my issue than I will be in touch

Best regards,
Bora Hoşadam

closed 2018-09-15 by BillyBlaze

BillyBlaze, 2018-09-15:
This has been released in 0.3.12

commit ce0fa109 (2018-09-01): :art: Disable webcamTab when active

Fixes #207
commit d2ca3b0f (2018-09-15): :art: Disable webcamTab when active

Fixes #207