BillyBlaze/OctoPrint-TouchUI #209: touchscreen does not launch touchui

ianjolin, 2018-04-19:
I have a fresh install of v138. The first plugin I installed was touchui, then I installed the pi desktop environment via cmd line and it does not seem to launch either via web access or the desktop. 
When I access the UI via another machine the TouchUI gui is accessible and operable but its not present and does not automatically launch on the pi touchscreen.
Are there test that I can run or area I can check to determine why this may not be launching?

BillyBlaze, 2018-05-08:
Sorry for the long delay, I am unable to look at this and will be looking into this as soon as I am able to.

deejayspinz, 2018-06-01:
+1.  I have this same issue. Followed the instructions. Did a fresh install of OctoPi, then followed instructions in the wiki to install TouchUI. Also installed LCD-show.  The TFT shows all the boot cmd line stuff as the Pi starts up.  When done, it shows "octopi login:" and does not boot to TouchUI.    I am using the Banggood 3.5" TFT.

BillyBlaze, 2018-09-02:
Hi @ianjolin,

Sorry for the delayed response. Could you perhaps share the contents of `/var/log/Xorg.0.log`?

@PlastiBots 
Thanks for sharing that!!

ianjolin, 2018-09-04:
Here is my log file.

 

The autostart doesn’t launch on my touchscreen.

BillyBlaze, 2018-09-04:
@ianjolin, I don't see anything.

ianjolin, 2018-09-04:
You didn’t receive an attached file?

BillyBlaze, 2018-09-04:
Nope, there are no files attached to your comment.

ianjolin, 2018-09-04:
Here is the contents of the log

 

[     6.095] 

X.Org X Server 1.19.2

Release Date: 2017-03-02

[     6.095] X Protocol Version 11, Revision 0

[     6.095] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian

[     6.095] Current Operating System: Linux octopi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l

[     6.095] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=266d4ded-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

[     6.096] Build Date: 18 October 2017  04:55:30PM

[     6.096] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 

[     6.096] Current version of pixman: 0.34.0

[     6.096]            Before reporting problems, check http://wiki.x.org

                to make sure that you have the latest version.

[     6.096] Markers: (--) probed, (**) from config file, (==) default setting,

                (++) from command line, (!!) notice, (II) informational,

                (WW) warning, (EE) error, (NI) not implemented, (??) unknown.

[     6.096] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep  4 12:45:27 2018

[     6.109] (==) Using system config directory "/usr/share/X11/xorg.conf.d"

[     6.124] (==) No Layout section.  Using the first Screen section.

[     6.125] (==) No screen section available. Using defaults.

[     6.125] (**) |-->Screen "Default Screen Section" (0)

[     6.125] (**) |   |-->Monitor "<default monitor>"

[     6.134] (==) No device specified for screen "Default Screen Section".

                Using the first device section listed.

[     6.134] (**) |   |-->Device "touchscreen"

[     6.134] (**) |   |-->GPUDevice "Allwinner A10/A13 FBDEV"

[     6.134] (==) No monitor specified for screen "Default Screen Section".

                Using a default monitor configuration.

[     6.134] (==) Automatically adding devices

[     6.135] (==) Automatically enabling devices

[     6.135] (==) Automatically adding GPU devices

[     6.146] (==) Max clients allowed: 256, resource mask: 0x1fffff

[     6.177] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.

[     6.177]            Entry deleted from font path.

[... 306 lines cut ...]
[     8.876] (II) No input driver specified, ignoring this device.

[     8.877] (II) This device may have been added with another device file.

[     8.906] (**) Telink Wireless Receiver: Applying InputClass "libinput pointer catchall"

[     8.906] (**) Telink Wireless Receiver: Applying InputClass "libinput keyboard catchall"

[     8.906] (II) Using input driver 'libinput' for 'Telink Wireless Receiver'

[     8.906] (**) Telink Wireless Receiver: always reports core events

[     8.906] (**) Option "Device" "/dev/input/event0"

[     8.906] (**) Option "_source" "_driver/libinput"

[     8.907] (II) libinput: Telink Wireless Receiver: is a virtual subdevice

[     8.907] (**) Option "config_info" "udev:/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:248A:FF0F.0001/input/input0/event0"

[     8.907] (II) XINPUT: Adding extended input device "Telink Wireless Receiver" (type: KEYBOARD, id 9)

[     8.907] (**) Option "xkb_model" "pc105"

[     8.907] (**) Option "xkb_layout" "gb"

BillyBlaze, 2018-09-12:
That looks okay,

What if you try this?
1. stop touchui `sudo service touchui stop`
1. run `sudo xinit ~/TouchUI-autostart/chromium.xinit`
1. post output here

closed 2018-10-03 by stale[bot]

Drakogol, 2019-02-10:
Im trying to restart this, as im having the same problem and can´t find a solution.
I tried as you suggested and this was the outcome:

pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit
[sudo] password for pi:

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20                   GMT 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight                  =600 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000                    dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=e100d113-02                   rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sun Feb 10 10:00:17 2019
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional info                  rmation.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

any ideas?

Drakogol, 2019-02-10:
In another thread i saw some solution and tried this:

pi@octopi:~ $ wget -qO - http://bintray.com/user/downloadSubjectPublicKey?username=bintray | sudo apt-key add -
OK
pi@octopi:~ $ echo "deb http://dl.bintray.com/kusti8/chromium-rpi jessie main " | sudo tee -a /etc/apt/sources.list
deb http://dl.bintray.com/kusti8/chromium-rpi jessie main
pi@octopi:~ $ sudo apt-get update
E: Malformed entry 4 in list file /etc/apt/sources.list (Component)
E: The list of sources could not be read.