BillyBlaze/OctoPrint-TouchUI #213: Webcam stops functioning after LCD and/or Touch UI installation

moveslikeyoda, 2018-05-02:
Having issue on RPi3b.  I installed in this order Octopi (from Octopi provided image), then webcam, which functioned fine, then Waveshare 3.5 LCD, then Touch UI.  After installing Touch UI, I not get the webcam stream.  I have tried the troubleshooting section addresses and neither work for me.  Performed sudo apt-get update as some digging suggested to no avail
[image]

BillyBlaze, 2018-05-08:
Sorry for the long delay, I am unable to look at this and will be looking into this as soon as I am able to.

moveslikeyoda, 2018-05-23:
Any luck?  I will add that after further testing, the webcam stream is only failing on the local (server) device.  The stream is still good on remote Octopi Web interface logins (e.g. PC browser). Just for some additional information, I am now running Octoprint 1.3.8 Octopi 0.15 and using the Raspi Cam

BillyBlaze, 2018-09-02:
Thanks for reporting, I have fixed this issue in the bootloader and if you update to the latest version that this will be fixed and you can use relative paths again.

To update your bootloader follow these steps:
1. Goto your TouchUI startup folder:

   ````
   cd ~/TouchUI-autostart/
   ````

1. Update the files with git*
   ````
   git pull
   ````

1. Restart TouchUI

   ````
   sudo service touchui restart
   ````

closed 2018-09-02 by BillyBlaze