BillyBlaze/OctoPrint-TouchUI #215: TouchUI fails to start on Octopi 0.15

qpens, 2018-05-04:
Hi, 
TouchUI will not boot I'm stuck in the terminal with the login option

I used instructions from this link 
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

That command fails
sudo sed -i 's/allowed_users=console/allowed_users=anybody/' /etc/X11/Xwrapper.config

Any solutions?

BillyBlaze, 2018-09-02:
Hi @qpens,

Sorry for the delayed response. Could you perhaps share the contents of /var/log/Xorg.0.log?

Also the command that fails can be ignored.

qpens, 2018-09-04:
[    10.199] 
X.Org X Server 1.19.2
Release Date: 2017-03-02
[    10.199] X Protocol Version 11, Revision 0
[    10.199] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
[    10.199] Current Operating System: Linux octopi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l
[    10.199] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=a456dfbf-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    10.199] Build Date: 18 October 2017  04:55:30PM
[    10.199] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 
[    10.199] Current version of pixman: 0.34.0
[    10.199] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    10.199] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    10.200] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Aug 18 07:40:15 2018
[    10.212] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    10.217] (==) No Layout section.  Using the first Screen section.
[    10.217] (==) No screen section available. Using defaults.
[    10.217] (**) |-->Screen "Default Screen Section" (0)
[    10.218] (**) |   |-->Monitor "<default monitor>"
[    10.224] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    10.224] (**) |   |-->Device "touchscreen"
[    10.224] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    10.224] (==) Automatically adding devices
[    10.224] (==) Automatically enabling devices
[    10.224] (==) Automatically adding GPU devices
[    10.237] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    10.239] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.239] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.239] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.239] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.239] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.239] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    10.239] 	Entry deleted from font path.
[    10.240] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    10.240] 	Entry deleted from font path.
[    10.240] (==) FontPath set to:
	built-ins
[    10.240] (==) ModulePath set to "/usr/lib/xorg/modules"
[    10.240] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    10.242] (II) Loader magic: 0x1fbf40
[    10.242] (II) Module ABI versions:
[    10.242] 	X.Org ANSI C Emulation: 0.4
[    10.242] 	X.Org Video Driver: 23.0
[    10.242] 	X.Org XInput driver : 24.1
[    10.242] 	X.Org Server Extension : 10.0
[    10.245] (--) using VT number 2

[    10.246] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    10.246] (II) no primary bus or device found
[    10.248] (II) LoadModule: "glx"
[    10.279] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    10.446] (II) Module glx: vendor="X.Org Foundation"
[    10.446] 	compiled for 1.19.2, module version = 1.0.0
[    10.446] 	ABI class: X.Org Server Extension, version 10.0
[    10.447] (II) LoadModule: "fbdev"
[    10.447] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    10.451] (II) Module fbdev: vendor="X.Org Foundation"
[    10.451] 	compiled for 1.19.0, module version = 0.4.4
[    10.451] 	Module class: X.Org Video Driver
[    10.451] 	ABI class: X.Org Video Driver, version 23.0
[    10.451] (II) FBDEV: driver for framebuffer: fbdev
[    10.470] (WW) Falling back to old probe method for fbdev
[    10.470] (II) Loading sub module "fbdevhw"
[    10.470] (II) LoadModule: "fbdevhw"
[    10.471] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    10.478] (II) Module fbdevhw: vendor="X.Org Foundation"
[    10.478] 	compiled for 1.19.2, module version = 0.0.2
[    10.478] 	ABI class: X.Org Video Driver, version 23.0
[    10.478] (EE) open /dev/fb1: No such file or directory
[    10.479] (EE) No devices detected.
[    10.479] (EE) 
Fatal server error:
[    10.479] (EE) no screens found(EE) 
[    10.479] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[    10.479] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    10.479] (EE) 
[    10.517] (EE) Server terminated with error (1). Closing log file.

BillyBlaze, 2018-09-04:
`[ 10.478] (EE) open /dev/fb1: No such file or directory`

Run `sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf`
Change `fb1` into `fb0`
Save (CTR-O, enter, CTR-X)

qpens, 2018-09-08:
so for one-second mouse cursor was shown on the screen after that again terminal

BillyBlaze, 2018-09-12:
Can you reshare `/var/log/Xorg.0.log`?

qpens, 2018-09-17:
X.Org X Server 1.19.2
Release Date: 2017-03-02
[    13.886] X Protocol Version 11, Revision 0
[    13.886] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
[    13.886] Current Operating System: Linux octopi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l
[    13.886] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=a456dfbf-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    13.886] Build Date: 18 October 2017  04:55:30PM
[    13.886] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 
[    13.886] Current version of pixman: 0.34.0
[    13.886] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    13.886] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    13.887] (==) Log file: "/var/log/Xorg.0.log", Time: Mon Sep 17 20:49:54 2018
[    13.930] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    13.939] (==) No Layout section.  Using the first Screen section.
[    13.939] (==) No screen section available. Using defaults.
[    13.939] (**) |-->Screen "Default Screen Section" (0)
[    13.939] (**) |   |-->Monitor "<default monitor>"
[    13.950] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    13.951] (**) |   |-->Device "touchscreen"
[    13.951] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    13.951] (==) Automatically adding devices
[    13.951] (==) Automatically enabling devices
[    13.951] (==) Automatically adding GPU devices
[    13.963] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    13.972] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.972] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.972] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.972] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.972] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.972] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    13.972] 	Entry deleted from font path.
[    13.973] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    13.973] 	Entry deleted from font path.
[    13.973] (==) FontPath set to:
	built-ins
[    13.973] (==) ModulePath set to "/usr/lib/xorg/modules"
[    13.973] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    13.982] (II) Loader magic: 0x1fbf40
[    13.982] (II) Module ABI versions:
[    13.982] 	X.Org ANSI C Emulation: 0.4
[    13.982] 	X.Org Video Driver: 23.0
[    13.983] 	X.Org XInput driver : 24.1
[    13.983] 	X.Org Server Extension : 10.0
[    13.986] (--) using VT number 2

[    13.986] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    13.987] (II) no primary bus or device found
[    13.990] (II) LoadModule: "glx"
[    14.023] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    14.178] (II) Module glx: vendor="X.Org Foundation"
[    14.178] 	compiled for 1.19.2, module version = 1.0.0
[    14.178] 	ABI class: X.Org Server Extension, version 10.0
[    14.179] (II) LoadModule: "fbdev"
[    14.179] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    14.195] (II) Module fbdev: vendor="X.Org Foundation"
[    14.195] 	compiled for 1.19.0, module version = 0.4.4
[    14.195] 	Module class: X.Org Video Driver
[    14.195] 	ABI class: X.Org Video Driver, version 23.0
[    14.195] (II) FBDEV: driver for framebuffer: fbdev
[... 62 lines cut ...]
[    14.821] (**) Option "xkb_layout" "gb"
[    14.968] (II) input device 'FULL HD 1080P Webcam: FULL HD 1', /dev/input/event0 is tagged by udev as: Keyboard
[    14.968] (II) input device 'FULL HD 1080P Webcam: FULL HD 1', /dev/input/event0 is a keyboard
[    14.974] (II) config/udev: Adding input device WaveShare WS170120 (/dev/input/event1)
[    14.974] (**) WaveShare WS170120: Applying InputClass "libinput touchscreen catchall"
[    14.974] (II) Using input driver 'libinput' for 'WaveShare WS170120'
[    14.974] (**) WaveShare WS170120: always reports core events
[    14.974] (**) Option "Device" "/dev/input/event1"
[    14.974] (**) Option "_source" "server/udev"
[    14.978] (II) input device 'WaveShare WS170120', /dev/input/event1 is tagged by udev as: Touchscreen
[    14.979] (II) input device 'WaveShare WS170120', /dev/input/event1 is a touch device
[    15.070] (**) Option "config_info" "udev:/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:0EEF:0005.0001/input/input1/event1"
[    15.070] (II) XINPUT: Adding extended input device "WaveShare WS170120" (type: TOUCHSCREEN, id 7)
[    15.071] (**) Option "AccelerationScheme" "none"
[    15.071] (**) WaveShare WS170120: (accel) selected scheme none/0
[... cut ...]
[    15.075] (II) input device 'WaveShare WS170120', /dev/input/event1 is tagged by udev as: Touchscreen
[    15.076] (II) input device 'WaveShare WS170120', /dev/input/event1 is a touch device
[    15.081] (II) config/udev: Adding input device WaveShare WS170120 (/dev/input/mouse0)
[    15.081] (II) No input driver specified, ignoring this device.
[    15.081] (II) This device may have been added with another device file.
[    20.054] (II) UnloadModule: "libinput"
[    20.056] (II) UnloadModule: "libinput"
[    20.148] (II) Server terminated successfully (0). Closing log file.

BillyBlaze, 2018-09-17:
I cant see anything wrong.

What is the output if you run these commands?

1. `sudo service touchui stop`
1. `sudo xinit ~/TouchUI-autostart/chromium.xinit`

qpens, 2018-09-17:
BillyBlaze please ignore that issue. Just updated to latest version and problem is solved now.
Thanks

BillyBlaze, 2018-09-17:
Okay glad you figured it out! Enjoy!

closed 2018-09-17 by BillyBlaze