BillyBlaze/OctoPrint-TouchUI #217: WebCam is not loading locally. Works fine on LAN/WAN.

bolsoncerrado, 2018-05-30:
Brand new, fresh, OctoPi 0.15.1+OctoPrint 1.38.0 SD image.

Everything works fine, but as title says, I get "Webcam stream not found" error ONLY in the local touchui "official" touch display of the Raspi.

BUT accessing Octoprint on the LAN or WAN shows the webcam stream just fine. 

What am I missing?!

Thanks!

Wyseman76, 2018-06-04:
I'm having the exact same issue. Using the 3B+ and the official 7" touch screen. I can connect via network and see the webcam, but on the local touchscreen I get the "Webcam stream not loaded" message.

bolsoncerrado, 2018-06-04:
Out of curiosity, do you have the snapstream plugin installed?

Wyseman76, 2018-06-04:
No I do not have the snapstream plugin installed.

bolsoncerrado, 2018-06-04:
Neither do I but its the only difference aside from Pi3 vs Pi3+ i have with other Pis with the cam working fine on local :?

moveslikeyoda, 2018-06-13:
I posted the same issue over a month ago, not sure anyone is getting responses right now.

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/213

BillyBlaze, 2018-06-13:
Sorry, I am unable to make a new release due to moving to a new house. Once I have a computer and internet setup then I am able to make a new release.

bolsoncerrado, 2018-07-16:
> Hi, I don't think this is a WAN/LAN issue the testing carried out is below
>
> I have
> Octoprint 1.3.8
> Octopi 0.15.1
> Touch UI 0.3.11
>
> Running on a Raspberry Pi 3B+ with official touchscreen
> Touchscreen does not display webcam when connected to WIFI (LAN) using
> TouchUI
> Desktop PC displays webcam when connected to WIFI(LAN) using Chrome No
> TouchUI
> Mobile Phone displays webcam when connected to WIFI (LAN) when using
> TouchUI
> Mobile Phone displays webcam when connected to 4G (WAN) when using TouchUI
>
> Therefore I don't think this is a WAN/LAN issue. It appears that the
> Webcam is not being displayed on the TouchUI Locally on the RaspberryPI.
> Possibly a browser issue

It's related to the Pi3B+ too, Idk why. My farm runs Pi3 mostly and all
display PiCams fine locally. The only one that runs on the B3+ does not.

jraydoran, 2018-07-27:
I too am having the same issue.

Using below for stream URL below does resolve for local touch UI view, but then does not work on WAN.
http://YOUR_IP:8080/?action=stream

Thank you!

BillyBlaze, 2018-09-02:
Thanks for reporting, I have fixed this issue in the bootloader and if you update to the latest version that this will be fixed and you can use relative paths again.

To update your bootloader follow these steps:
1. Goto your TouchUI startup folder:

   ````
   cd ~/TouchUI-autostart/
   ````

1. Update the files with git*
   ````
   git pull
   ````

1. Restart TouchUI

   ````
   sudo service touchui restart
   ````

closed 2018-09-02 by BillyBlaze