BillyBlaze/OctoPrint-TouchUI #221: OctoPi 1.3.9rc1 Release Candidate

tibmeister, 2018-06-24:
So I went ahead and upgraded my OctoPrint to this version, which has a new Tornado version, and now TouchUI is "broke".  Here's what I've been able to glean from the developer console of Chrome when starting TouchUI not in kiosk mode.  Maybe it's helpful???
I'm running on a rPI Model 3 b+ with Raspian Stretch.

[image]
[image]

tibmeister, 2018-06-24:
So the workaround I am using is changing the chrome args from

`CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
                $(dirname "$0")/load-screen/startup.html"`

to

`CHROME_ARGS="   --no-first-run --kiosk #$TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
                $(dirname "$0")/load-screen/startup.html"`

Seems to work for now.

acrespog, 2018-06-24:
@tibmeister what have you changed in CHROME_ARGS? 
I didn't see any diference.

Thanks.

acrespog, 2018-06-24:
Thanks! I have checked ten times without success.

macv36, 2018-07-27:
Hi,
After updating OctoPrint to Version 1.3.9, got RSOD (Red Screen of Dead)
I tried:
#$TOUCHUI_ARGS $TOUCHUI_DEBUG
but this stops TouchUI from loading on the Pi. I can manage Octoprint from PC, but I like to also have the advantage of not having to go back to the PC if I can change something directly on the Pi.
I hope a solution is on the works soon.

dwcrall, 2018-09-02:
I loaded TouchUI this morning, and it was working fine.  I tried to take the update and I it hangs at "Loading TouchUI".  It appears to be the same problem discussed in this thread.  Is there a way to roll back the rev?  Is there a special place to get the fixed version?

BillyBlaze, 2018-09-02:
You can uninstall TouchUI and install it again with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

That should fix it.

macv36, 2018-09-03:
I can confirm the update fixed my issue. Thanks!

rhys100, 2018-09-03:
fixed for me! cheers. Might be good to push this out now, as i did a brand new setup today and wasted 45 mins trying to work out why it wouldnt load :D

jraydoran, 2018-09-03:
Working for me too. Thank you for your effort. Appreciated.

otherguy, 2018-09-03:
Fixed it for me too! Thank you for the effort, @BillyBlaze!

jamskate, 2018-09-03:
Yep, Fixed it for me too..  Thank you!!!!

maganzi, 2018-09-06:
Hey, can someone explain to me how to download and install the maintenance zip archive to my raspberry?

tigrrr, 2018-09-07:
Just download the file and install it with Octoprint Plugin manager.

@BillyBlaze, I've installed the TouchUI maintenance version, but I still get the RSOD.

maganzi, 2018-09-07:
Thank you Tigrrr!
My HDMI Display runs perfectly but TouchUI doesn´t start automatically... i tried so much =/ i think the new build is the problem

jraydoran, 2018-09-07:
Although it initially worked for me. It stopped working after I updated to the latest bootloader.

maganzi, 2018-09-07:
Which is the way to autostart this build of TouchUI?

BillyBlaze, 2018-09-08:
@tigrrr, @jraydoran
Thanks for reporting, could you provide me a list of plugins installed?

@maganzi 
It sounds like you have other issues then what is reporting in this issue, please create a new issue for help.

tigrrr, 2018-09-08:
hi, here's my plugin list: 
Action Command Prompt Support  
Action Commands (0.1)
Announcement Plugin  
Bed Leveling Wizard (0.1.6)
Core Wizard  
CuraEngine (<= 15.04)  
Discovery  
Emergency Stop Button (0.1.2)
Fullscreen Plugin (0.0.4)
Logging  
Octolapse (0.3.1)
OctoPi Support Plugin  
Plugin Manager  
Printer Safety Check  
Slicer (1.4.0)
Software Update  
Touchtest Plugin (0.1.1)
TouchUI (0.3.12)
Virtual Printer  

I'm gonna retry a fresh install next week with TouchUI only and report.

eboston, 2018-09-08:
Instead of doing a fresh install, try disabling all the other plug-ins 
first.  See if that has any effect.

BillyBlaze, 2018-09-09:
@tigrrr,
Thanks for the list, I was able to reproduce your issue with "Bed Leveling Wizard (0.1.6)" enabled.

@jraydoran,
Do you have this plugin installed too?

jraydoran, 2018-09-09:
Hi,

Here is my list of plugins. I do not have this plugin installed.

Action Command Prompt Support
Announcement Plugin
AstroPrint (1.1.0)
Core Wizard
CuraEngine (<= 15.04)
Discovery
Emergency Stop Button (0.1.2)
Logging
MQTT (0.7.1)
MQTT Publish (0.5.0)
Octolapse (0.3.1)
OctoPi Support Plugin
Plugin Manager
PolarCloud (1.6)
Printer Safety Check
Pushover (0.2.1)
Software Update
Tasmota-MQTT (0.3.1)
TouchUI (0.3.12)
Virtual Printer

Thanks for the effort!

Kind Regards

BillyBlaze, 2018-09-09:
@jraydoran 

Thanks for reporting, I cannot reproduce your issue at the moment, could you install the latest maintenance branch, update the bootloader and create a new issue with the error that is being shown in the bootloader?

jraydoran, 2018-09-09:
100%, Thank you!

BillyBlaze, 2018-09-11:
@jraydoran any updates? I want to release but I like to resolve your issue too before releasing.

jraydoran, 2018-09-11:
Updated bootloader and installed latest maintenance release.
Disabled all plugins and still getting “Connecting to TouchUI” with a counter to the left that counts to 30 and then fails.

maganzi, 2018-09-11:
> Updated bootloader and installed latest maintenance release.
> Disabled all plugins and still getting “Connecting to TouchUI” with a counter to the left that counts to 30 and then fails.

me2

BillyBlaze, 2018-09-12:
Thanks for the info, can you provide me your octoprint.log?

jraydoran, 2018-09-12:
Thanks for your effort.

See attached

[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2375186/octoprint.log)

jraydoran, 2018-09-13:
@BillyBlaze, I did update with the same results.

BillyBlaze, 2018-09-13:
@jraydoran Thanks for the feedback, can we continue this in #239 ?

commit 4a73bb1e (2018-09-15): :bug: Fix Bed Leveling Wizard issues

Fixes #221
closed 2018-09-15 by BillyBlaze