BillyBlaze/OctoPrint-TouchUI #222: OctoPi 1.3.9rc1 Release Candidate login failed and error console

devildant, 2018-06-25:
hello
i have an error inside the console
`packed_core.js?458964a9:14356 Error calling onAfterTabChange on view model TemperatureViewModel : Error: Invalid dimensions for plot, width = 0, height = 0
    at Canvas.resize (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:16778:10)
    at new Canvas (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:16757:8)
    at setupCanvases (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:17956:23)
    at new Plot (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:17350:9)
    at Function.$.plot (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:19789:20)
    at TemperatureViewModel.self._initializePlot (http://192.168.1.2:5000/static/webassets/packed_core.js?458964a9:5380:27)
    at TemperatureViewModel.self.onAfterTabChange (http://192.168.1.2:5000/static/webassets/packed_core.js?458964a9:5757:22)
    at callViewModelIf (http://192.168.1.2:5000/static/webassets/packed_core.js?458964a9:14343:35)
    at http://192.168.1.2:5000/static/webassets/packed_core.js?458964a9:14292:13
    at Pn (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:11208:530)
callViewModelIf @ packed_core.js?458964a9:14356
(anonymous) @ packed_core.js?458964a9:14292
Pn @ packed_libs.js?331a1bb5:11208
(anonymous) @ packed_libs.js?331a1bb5:11229
callViewModelsIf @ packed_core.js?458964a9:14290
callViewModels @ packed_core.js?458964a9:14284
onAfterTabChange @ packed_core.js?458964a9:14943
(anonymous) @ packed_core.js?458964a9:14965
mightThrow @ packed_libs.js?331a1bb5:3535
process @ packed_libs.js?331a1bb5:3603
setTimeout (async)
(anonymous) @ packed_libs.js?331a1bb5:3641
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
fire @ packed_libs.js?331a1bb5:3407
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
ready @ packed_libs.js?331a1bb5:3879
completed @ packed_libs.js?331a1bb5:3889
`

and when I try to log in via the menu, and when I touch the login menu, the panel closes and can not be logged

i have an error : 
`Error calling onStartupComplete on view model TouchUI : TypeError: url.indexOf is not a function
    at jQuery.fn.init.jQuery.fn.load (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:9858:13)
    at g (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:6582:558)
    at f (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:6582:265)
    at new b (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:6582:2706)
    at HTMLDivElement.<anonymous> (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:6582:2984)
    at Function.each (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:355:19)
    at jQuery.fn.init.each (http://192.168.1.2:5000/static/webassets/packed_libs.js?331a1bb5:190:17)
    at jQuery.fn.init.a.fn.(anonymous function) [as tinycolorpicker] (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:6582:2918)
    at TouchUI.knockout.isReady (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:7914:29)
    at TouchUI.self.onStartupComplete (http://192.168.1.2:5000/static/webassets/packed_plugins.js?464274d0:8004:33)
callViewModelIf @ packed_core.js?458964a9:14356
(anonymous) @ packed_core.js?458964a9:14292
Pn @ packed_libs.js?331a1bb5:11208
(anonymous) @ packed_libs.js?331a1bb5:11229
callViewModelsIf @ packed_core.js?458964a9:14290
callViewModels @ packed_core.js?458964a9:14284
bindViewModels @ packed_core.js?458964a9:15096
setTimeout (async)
(anonymous) @ packed_core.js?458964a9:15135
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
done @ packed_libs.js?331a1bb5:9306
(anonymous) @ packed_libs.js?331a1bb5:9549
load (async)
send @ packed_libs.js?331a1bb5:9568
ajax @ packed_libs.js?331a1bb5:9207
OctoPrintClient.ajax @ packed_client.js?4a844250:143
OctoPrintClient.get @ packed_client.js?4a844250:156
OctoPrintSettingsClient.get @ packed_client.js?4a844250:1236
SettingsViewModel.self.requestData @ packed_core.js?458964a9:4122
fetchSettings @ packed_core.js?458964a9:15122
setTimeout (async)
(anonymous) @ packed_core.js?458964a9:15192
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
done @ packed_libs.js?331a1bb5:9306
(anonymous) @ packed_libs.js?331a1bb5:9549
load (async)
send @ packed_libs.js?331a1bb5:9568
ajax @ packed_libs.js?331a1bb5:9207
OctoPrintClient.ajax @ packed_client.js?4a844250:143
OctoPrintClient.ajaxWithData @ packed_client.js?4a844250:152
OctoPrintClient.post @ packed_client.js?4a844250:164
OctoPrintClient.postJson @ packed_client.js?4a844250:177
OctoPrintBrowserClient.passiveLogin @ packed_client.js?4a844250:551
LoginStateViewModel.self.requestData @ packed_core.js?458964a9:2311
onServerConnect @ packed_core.js?458964a9:15172
(anonymous) @ packed_core.js?458964a9:15190
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
deferred.(anonymous function) @ packed_libs.js?331a1bb5:3737
DataUpdater.self._onConnectMessage @ packed_core.js?458964a9:12905
(anonymous) @ packed_client.js?4a844250:388
Pn @ packed_libs.js?331a1bb5:11208
(anonymous) @ packed_libs.js?331a1bb5:11229
OctoPrintSocketClient.propagateMessage @ packed_client.js?4a844250:387
(anonymous) @ packed_client.js?4a844250:478
(anonymous) @ packed_libs.js?331a1bb5:11225
_t @ packed_libs.js?331a1bb5:11213
(anonymous) @ packed_libs.js?331a1bb5:11224
(anonymous) @ packed_libs.js?331a1bb5:11229
onMessage @ packed_client.js?4a844250:477
EventTarget.dispatchEvent @ packed_libs.js?331a1bb5:22527
(anonymous) @ packed_libs.js?331a1bb5:23240
SockJS._transportMessage @ packed_libs.js?331a1bb5:23238
EventEmitter.emit @ packed_libs.js?331a1bb5:22443
WebSocketTransport.ws.onmessage @ packed_libs.js?331a1bb5:25314
`

devildant, 2018-07-09:
no body ?

BillyBlaze, 2018-07-10:
Hi @devildant,

There is already a report of this issue on #221 it also has a temporary fix. Once I am done with labour on my new house I will fix this better.

closed 2018-07-10 by BillyBlaze