BillyBlaze/OctoPrint-TouchUI #223: TouchUI not working after upgrade to 1.3.9 RC2

sajovicd, 2018-07-09:
Hello,
I am have OctoPrint 1.3.8 running on OctoPi 0.13.0 and 7" touch screen and TouchUI on it.
Everything was working OK until upgrade to 1.3.9 RC2 (RC1 was OK). 
The screen is stuck at connecting to octopi.
Downgrading to 1.3.8 and it is working again.

Richardgriff, 2018-07-17:
Confirmed. I get the same issue

Phife54, 2018-07-26:
Octoprint 1.3.9 was released and now TouchUI doesnt work.

tkardaridis, 2018-07-26:
Confrimed. Octoprint 1.3.9 was released and now TouchUI doesnt work. I just upgrade to 1.3.9 and touchui remains on loading screen.

jraydoran, 2018-07-27:
Confirmed. I too get the same issue.
Thank you!

synman, 2018-07-28:
@BillyBlaze glad you're looking into it.  I just pushed to 1.3.9 general and experienced the same problem.  touchui is failing within chromium on startup.

The only other change besides 1.3.9 I've made is installing Octolapse.  Haven't bothered disabling it as others mentioned the 1.3.9 RC but I can if simply hasn't been disclosed yet.

Holler if you need help debugging.... otherwise I'll just hang tight.

leitenky, 2018-07-28:
Hi I too am having the same issue. I have also noticed if I have touchUI enabled for low resolution devices automatically I cannot connect to octoprint on my local network. However if I port forward through my router externally I can access touchUI and disable this setting I can now atleast log into my octoprint from a PC. Not sure if this is all related some how

NedSin, 2018-07-28:
Hello, 
i try with complete frsh install on rpi 3b ... 

Same Problem, touch ui stays in loading screen ...

armysolo, 2018-07-29:
same issue with a 7 inch touch screen. I can still connect through my PC but cannot through the pi. the screen says connecting to TouchUI failed. No error message

mantisrobot, 2018-07-30:
Same here. 1.3.9 breaks TouchUI on both a 3.5" screen and official 7" pi screen.

tre4b, 2018-07-30:
I did a full re-install of the lot and installed touchui again following your instructions to set it up as default.  All is now working with the caveat that the touch is back to front (up/down)=(left/right) and visa versa.  Clearly something changed that a full reinstall could almost but not quite fix.

amoose136, 2018-08-04:
I don’t think it has anything to do with screen size. I can load it on my phone but can’t on the official 7” cap touch screen directly connected to the pi.

intrglctcrevfnk, 2018-08-13:
Same here- waited 18 days to update to 1.3.9 but failed to see this bug.  Works fine on normal webpage view but not the touchUI interface on my raspberry pi.

MrArmageddon, 2018-08-24:
Also experiencing "**TouchUI Startup Failed..**." after updating OctoPrint to v1.3.9.  Using the 7in RPi Touchscreen.
[image]

Phife54, 2018-08-24:
Is there any updates on this? Is the developer working on a fix? Or are there other solutions we can try to fix this? What about a different touchscreen plugin? I too use the 7" Pi LCD and its been great, but now they are useless on my 2 machines.

BillyBlaze, 2018-09-01:
Thanks everyone for the feedback, I am currently working on this issue and plan to release a.s.a.p.

However this issue is a similar to #221 so therefor I am closing this.

closed 2018-09-01 by BillyBlaze

mchaos, 2018-09-11:
Any fix? It seems like when touchui fails, it knocks out my wifi as well

synman, 2018-09-11:
@mchaos  I doubt the two are related.  I've been printing with no issues since installing 1.3.9 ... I just have the RSOD on my touchscreen.

mchaos, 2018-09-12:
if it's not touch UI causing my network just disconnecting issues and how
come when I kill all the processes for touch UI everything's fine

BillyBlaze, 2018-09-12:
@synman 
Your issue should be fixed on the maintenance branch. You can uninstall TouchUI and install it again with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

@mchaos 
That sounds very weird, TouchUI does not launch or tinker with any system settings/programs, It only launches Chromium with Xserver. Both shouldn't affect your Wifi. However since you've another issue then what is beeing reported, I kindly ask you to create a new issue for this.

mchaos, 2018-09-12:
I think whatever is not working is causing something to hang. I went down a
rabbit hole last night. When touch UI would try to load I'd get a lightning
bolt and the CPU would throttle. Then the pi would crawl. So I installed a
buck converter and had to go to 5.7 volts to solve the issue. No more
throttling and I thought I fixed it but I would eventually have the issue.
Maybe it's chrome. I kill all chrome instances and it's solved.

synman, 2018-09-14:
> @synman
> Your issue should be fixed on the maintenance branch. You can uninstall TouchUI and install it again with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

Thank you @BillyBlaze -- back in business

B3N73, 2018-09-14:
I've just been installing TouchUI this morning after finding it on Adafruit.
Thanks to this forum, I've been able to sort out the problem stated by the OP.

Many thanks  @BillyBlaze for your great efforts! Keep up the good work.