BillyBlaze/OctoPrint-TouchUI #225: Can't boot to browser because of cross origin (cors) error

micdenny, 2018-07-21:
OctoPrint 1.3.8 running on OctoPi 0.15.1

chromium-browser:
  Installed: 65.0.3325.181-0+rpt4
  Candidate: 65.0.3325.181-0+rpt4

from chromium logs I see a row per retry like this:

> [3277:3277:0721/161720.199354:INFO:CONSOLE(0)] "Failed to load http://localhost:5000/plugin/touchui/ping: The 'Access-Control-Allow-Origin' header has a value 'null' that is not equal to the supplied origin. Origin 'null' is therefore not allowed access.", source: file:///home/pi/TouchUI-autostart/load-screen/startup.html (0)

if I check from my browser I see that the server reply this in response header:

Access-Control-Allow-Origin: *

any ideas?

micdenny, 2018-07-21:
I read that browsers like chrome are really restrictive on using Ajax calls from file browsing, so why not use a small web server to host that loading page?

I will give a try with python server that should be easy as run this command:

python -m SimpleHTTPServer

micdenny, 2018-07-22:
it worked great!!

just run a simple server with this command:

```
python -m SimpleHTTPServer > /dev/null 2>&1 &
```

then change the link to browse on `chromium.xinit`

from:

```
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
                $(dirname "$0")/load-screen/startup.html"
```

to:

```
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
                http://localhost:8000/startup.html"
```

simply change from `$(dirname "$0")/load-screen/startup.html"` to `http://localhost:8000/startup.html`

then I had also to change the `script.js` and set the `prefix` to `http://0.0.0.0/` otherwise the touchui load but it doesn't work great, the control page doesn't show up.

now it boots just fine, but I have to put the web server to startup before calling chrome.

micdenny, 2018-07-23:
I did a small POC and submitted as PR here: https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/pull/1

qpens, 2018-07-26:
Where can I find chromium.xinit?

micdenny, 2018-07-26:
> Where can I find chromium.xinit?

@qpens under `/home/pi/TouchUI-autostart/chromium.xinit`

micdenny, 2018-08-14:
the bot is having hard times on this issue 😂

BillyBlaze, 2018-09-01:
Hi @micdenny,

First off, sorry for the long delay, I was moving to a new house and neglected this project. I am currently working on a new release so I was going over all the issues.

Secondly, I love your workaround, could you perhaps do one more small thing? Could you make port 8000 configurable through the [default file](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/touchui.default) in case someone wants to use another port; it's easier to change that config.

closed 2018-09-15 by BillyBlaze

BillyBlaze, 2018-09-15:
@micdenny, I needed your script to add a cache-buster to the bootloader. I therefor accepted your PR and made the changes myself. Thanks again for your work!

micdenny, 2018-09-15:
Ok thank you @billyblaze and sorry for the delay.