BillyBlaze/OctoPrint-TouchUI #228: Can't load touch ui after octoprint 1.3.9 update

YaYaLanD, 2018-07-28:
Hello !

Since the latest update, i can't access the touch ui on my 3.5" screen, i get a failed to load in red.
Git pull tells me i have the latest version of touch ui.

I hop you will be able to solve this soon : )

Thank you for all your efforts !
Regards

IdefixRC, 2018-07-28:
same issue here. This was already mentioned in an earlier bug report though.
Linking it here #223

YaYaLanD, 2018-07-28:
yeah i saw the bug report #223 but as this # is related with RC2 of 1.3.9, i thought i might open another thread.

ram64, 2018-07-28:
Same. TouchUI works well on desktop, in Chrome browser but on my RPI B3+ with official touchscreen I get **Connecting to TouchUI failed...**

scotthay513, 2018-07-28:
Same here, was working well before the last update and now the error message upon loading TouchUI.  Octoprint is working fine however, I can still access it through the webserver.

Kremptor, 2018-07-28:
Same Problem, TouchUI does not start after update 1.3.9.

Kremptor, 2018-07-28:
I did a downgrade to 1.3.8 and TouchUI is working.

~/oprint/bin/pip install https://github.com/foosel/OctoPrint/archive/1.3.8.zip
sudo reboot

IdefixRC, 2018-07-29:
thanks. going back to 1.3.8 till this is fixed

bolsoncerrado, 2018-07-30:
Same here on Pi3B and Pi3B+

Kmarkopl, 2018-07-30:
Have same issue, PI3+ TFT 5" screen

Nappyy, 2018-08-02:
same here, pi3b+

Tronnic, 2018-08-04:
Same here, Pi3B

TisDone, 2018-08-04:
Another report, same problem. Pi 3b+ with the official Raspberry 7" Touchscreen.

thedragonlord71, 2018-08-05:
Another report, me too with the official Raspberry 7" Touchscreen

einfachTobi, 2018-08-13:
Have the same issue and got a workaround for this:
Edit ~/TouchUI-autostart/chromium.xinit:
`nano ~/TouchUI-autostart/chromium.xinit `

Change the variable CHROME_ARGS to:
`CHROME_VERSION=$(dpkg -s chromium-browser | grep "Version: " | sed -e 's/Version: //g')
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
		'http://localhost'"
`
(last line changed)
This will not load the startup-screen of TouchUI-autostart, but at least TouchUI will start.
Verify with a restart of the service or reboot of the pi
`sudo service touchui restart` or `sudo reboot`

bolsoncerrado, 2018-08-13:
Wow thanks for that ;)

bolsoncerrado, 2018-08-13:
I can confirm @einfachTobi 's solution works....

BUT NOW

I have an issue while trying to LOGIN using TouchUI. It won't do anything. I can touch any options on the menu, and move thru all the screens TouchUI offers for Octoprint, but it just refuses to show login fields :( ie nothing happens when I click Login except the menu disappearing, of course.

jraydoran, 2018-08-13:
@einfachTobi
Same issue as @bolsoncerrado. Cannot "login" and no "control" screen.

Kind Regards
Jason

bolsoncerrado, 2018-08-13:
@jraydoran  the "no control screen" is normal because you're not logged in.

jraydoran, 2018-08-14:
Makes sense. Thank you!!

einfachTobi, 2018-08-14:
I'm sorry i can't help with that because I don't use the login and dont't know how it works. Maybe you can find another method for authentification and deactivate the octoprint login as long as the underlying problem with TouchUI-autostart has been solved.

jonnyhifi, 2018-08-23:
yes ! worked for me, I never thought it would take 2 man days to configure a raspberry pi from scratch... but I guess kids can do it... ;)

jace055, 2018-08-26:
After using einfachTobi's technique above, reboot then deactivate the touchui plugin.  Reboot once again to the normal octoprint screen.  login and check the remember me box.  Reactive the touchui plugin and reboot one last time and you should be logged in and in the touch ui.

BillyBlaze, 2018-09-01:
Thanks everyone for the feedback, I am currently working on this issue and plan to release a.s.a.p.

However this issue is a similar to #221 so therefor I am closing this.

closed 2018-09-01 by BillyBlaze