BillyBlaze/OctoPrint-TouchUI #230: Can't login with TouchUI

dikodahan, 2018-08-08:
running the latest OctoPi with OctoPrint and the latest TouchUI, however, I am not able to login. From the menu icon other components respond, it is just the login the accept the "click" just not doing anything.

amoose136, 2018-08-08:
I noticed the same thing. Toggling “touchscreen mode” seems to sometimes make it work as a temporary work around.

thethinkingman, 2018-08-15:
I have the same issue after installing 1.3.9 and 0.15 from scratch and doing an install of the plugins.
I deactivate TouchUI and then can login using the desktop interface. Then I switch back.
I am using an iPhone 6s.

frioSC, 2018-08-15:
Same thing here.  Clicking login in TouchUI simply does nothing.  The work around @amoose136 mentions worked for me.

dikodahan, 2018-08-15:
Seems like this is not a one off and obviously not a minor issue. Can we please share an update on status to getting this fixed?

BillyBlaze, 2018-09-02:
Thanks all for the feedback. Could anybody perhaps install the maintenance branch to see if this issue still persists with the latest fixes?

You can uninstall TouchUI and install it with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

BillyBlaze, 2018-09-04:
Awesome, thanks for the confirmation!

BillyBlaze, 2018-09-10:
Thank you! Very soon. I am waiting for feedback on the last open issue.

BillyBlaze, 2018-09-15:
I am assuming this has been resolved, please create a new issue when the issue is coming back.

closed 2018-09-15 by BillyBlaze